nightscout/AndroidAPS #460: Automation: Connection to Bluetooth device causes AAPS to crash when selected

happygirlnyc70, 2021-04-05:
When setting up an automation task AAPS crashes if connection to bluetooth device is selected.
Pixel 4a 5G Android 11; AAPS 2.8.2 build 19454adc-2021-01.23-10:27

[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/6260918/AndroidAPS.log)

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

MilosKozak, 2021-06-14:
No sign of crash in log and I cannot replicate the issue in DEV branch .
Open new issue if you find more info

closed 2021-06-14 by MilosKozak

happygirlnyc70, 2021-06-14:
Hi Milos, I will rebuild the APK and reinstall on my phone.  Perhaps something is corrupted. Thanks

happygirlnyc70, 2021-06-15:
Hi Milos, I just wanted to update you. Yesterday when I expected to replicate the issue and look at the Android Bluetooth logs it worked.  My guess is that this was android issue, which has been resolved because I've recently updated the OS. Thanks again for your efforts.