nightscout/AndroidAPS #483: Issues when doing profile switch on DANA R

vmichalec, 2021-04-17:
I perform profile switch and it often leads to problems:

* Today I received more than five notifications that profile was updated. During that time connection status bar shows lot of communication with pump. I'm adding logs. You can check from 10:31:52.

* I have also older logs with same action but it was unable to properly connect and update basal in the pump.  Log from 14:00:10.

[Profile_switching_problems.zip](https://github.com/nightscout/AndroidAPS/files/6329463/Profile_switching_problems.zip)

My configuration: AAPS 2.8.2, Nightscout: 14.2.1, Samsung Galaxy S10 Lite, DANA R Model: 03, Protocol: 03, Code: B3. I didn't noticed other communication problems between pump and phone.

I see in logs also this part bellow - does it mean that multiple threads are trying to connect to pump?
```
14:00:28.567 [Thread-25153] D/PUMPQUEUE: [QueueThread.run():114]: connecting 17
14:00:28.569 [Thread-25154] D/PUMPQUEUE: [QueueThread.run():114]: connecting 17
14:00:28.570 [Thread-25157] D/PUMPQUEUE: [QueueThread.run():114]: connecting 17
14:00:28.570 [Thread-25156] D/PUMPQUEUE: [QueueThread.run():114]: connecting 17
14:00:28.571 [Thread-25155] D/PUMPQUEUE: [QueueThread.run():114]: connecting 17

14:00:29.570 [Thread-25153] D/PUMPQUEUE: [QueueThread.run():114]: connecting 18
14:00:29.571 [Thread-25157] D/PUMPQUEUE: [QueueThread.run():114]: connecting 18
14:00:29.572 [Thread-25154] D/PUMPQUEUE: [QueueThread.run():114]: connecting 18
14:00:29.572 [Thread-25156] D/PUMPQUEUE: [QueueThread.run():114]: connecting 18
14:00:29.573 [Thread-25155] D/PUMPQUEUE: [QueueThread.run():114]: connecting 18

14:00:30.573 [Thread-25156] D/PUMPQUEUE: [QueueThread.run():114]: connecting 19
14:00:30.573 [Thread-25157] D/PUMPQUEUE: [QueueThread.run():114]: connecting 19
14:00:30.573 [Thread-25153] D/PUMPQUEUE: [QueueThread.run():114]: connecting 19
14:00:30.573 [Thread-25155] D/PUMPQUEUE: [QueueThread.run():114]: connecting 19
14:00:30.574 [Thread-25154] D/PUMPQUEUE: [QueueThread.run():114]: connecting 19

14:00:31.575 [Thread-25156] D/PUMPQUEUE: [QueueThread.run():114]: connecting 20
14:00:31.575 [Thread-25157] D/PUMPQUEUE: [QueueThread.run():114]: connecting 20
14:00:31.575 [Thread-25155] D/PUMPQUEUE: [QueueThread.run():114]: connecting 20
14:00:31.575 [Thread-25153] D/PUMPQUEUE: [QueueThread.run():114]: connecting 20
14:00:31.575 [Thread-25154] D/PUMPQUEUE: [QueueThread.run():114]: connecting 20
```

Or maybe this? Each thread with own counter value (also in UI the working status bar jumps forward and back... connecting 30, 38, 31, 39, ...)
```
14:02:53.263 [Thread-25333] D/PUMPQUEUE: [QueueThread.run():114]: connecting 30
14:02:53.265 [Thread-25334] D/PUMPQUEUE: [QueueThread.run():114]: connecting 30
14:02:53.395 [Thread-25157] D/PUMPQUEUE: [QueueThread.run():114]: connecting 38
14:02:53.395 [Thread-25154] D/PUMPQUEUE: [QueueThread.run():114]: connecting 38
14:02:53.395 [Thread-25155] D/PUMPQUEUE: [QueueThread.run():114]: connecting 38
14:02:54.265 [Thread-25333] D/PUMPQUEUE: [QueueThread.run():114]: connecting 31
14:02:54.266 [Thread-25334] D/PUMPQUEUE: [QueueThread.run():114]: connecting 31
14:02:54.396 [Thread-25157] D/PUMPQUEUE: [QueueThread.run():114]: connecting 39
14:02:54.397 [Thread-25154] D/PUMPQUEUE: [QueueThread.run():114]: connecting 39
14:02:54.398 [Thread-25155] D/PUMPQUEUE: [QueueThread.run():114]: connecting 39
14:02:55.266 [Thread-25333] D/PUMPQUEUE: [QueueThread.run():114]: connecting 32
14:02:55.267 [Thread-25334] D/PUMPQUEUE: [QueueThread.run():114]: connecting 32
14:02:55.397 [Thread-25157] D/PUMPQUEUE: [QueueThread.run():114]: connecting 40
14:02:55.399 [Thread-25154] D/PUMPQUEUE: [QueueThread.run():114]: connecting 40
14:02:55.400 [Thread-25155] D/PUMPQUEUE: [QueueThread.run():114]: connecting 40
14:02:56.269 [Thread-25333] D/PUMPQUEUE: [QueueThread.run():114]: connecting 33
14:02:56.270 [Thread-25334] D/PUMPQUEUE: [QueueThread.run():114]: connecting 33
14:02:56.399 [Thread-25157] D/PUMPQUEUE: [QueueThread.run():114]: connecting 41
14:02:56.400 [Thread-25154] D/PUMPQUEUE: [QueueThread.run():114]: connecting 41
14:02:56.400 [Thread-25155] D/PUMPQUEUE: [QueueThread.run():114]: connecting 41
```

MilosKozak, 2021-04-18:
this is wrong. there should be only 1 thread running and connecting to the pump
it must be some underlaing problem causing this ie some crash before ....
definitely it's not safe to use AAPS as it is now
i'd recommend reinstall AAPS form scratch and import config and it not helps , factory reset the phone

vmichalec, 2021-04-22:
I reinstalled the app but doesn't helped at all.  Here are again some logs:

```
18:36:31.570 [RxCachedThreadScheduler-83] D/PROFILE: [CommandQueue$1.accept():114]: onProfileSwitch
18:36:31.571 [RxCachedThreadScheduler-1518] D/PROFILE: [CommandQueue$1.accept():114]: onProfileSwitch
18:36:31.570 [RxCachedThreadScheduler-82] D/DATATREATMENTS: [TreatmentsPlugin.initializeProfileSwitchData():232]: initializeProfileSwitchData
18:36:31.572 [RxCachedThreadScheduler-1519] D/PROFILE: [CommandQueue$1.accept():114]: onProfileSwitch
18:36:31.571 [RxCachedThreadScheduler-1518] D/PUMPQUEUE: [CommandQueue.setProfile():374]: Correct profile already set
18:36:31.572 [RxCachedThreadScheduler-1520] D/PROFILE: [CommandQueue$1.accept():114]: onProfileSwitch
18:36:31.573 [RxCachedThreadScheduler-83] D/PUMPQUEUE: [CommandQueue.setProfile():374]: Correct profile already set
18:36:31.571 [RxCachedThreadScheduler-765] D/PROFILE: [CommandQueue$1.accept():114]: onProfileSwitch
18:36:31.574 [RxCachedThreadScheduler-2609] D/PROFILE: [CommandQueue$1.accept():114]: onProfileSwitch
18:36:31.594 [RxCachedThreadScheduler-1520] D/PUMP: [AbstractDanaRPlugin.isThisProfileSet():186]: Diff found. Hour: 11 Pump: 0.9 Profile: 0.8
18:36:31.594 [RxCachedThreadScheduler-2609] D/PUMP: [AbstractDanaRPlugin.isThisProfileSet():186]: Diff found. Hour: 11 Pump: 0.9 Profile: 0.8
18:36:31.595 [RxCachedThreadScheduler-765] D/PUMP: [AbstractDanaRPlugin.isThisProfileSet():186]: Diff found. Hour: 11 Pump: 0.9 Profile: 0.8
18:36:31.595 [RxCachedThreadScheduler-1519] D/PUMP: [AbstractDanaRPlugin.isThisProfileSet():186]: Diff found. Hour: 11 Pump: 0.9 Profile: 0.8
18:36:31.595 [RxCachedThreadScheduler-1520] D/PUMPQUEUE: [CommandQueue.isThisProfileSet():561]: Current profile: {MY_PROFILE_DATA}
18:36:31.596 [RxCachedThreadScheduler-765] D/PUMPQUEUE: [CommandQueue.isThisProfileSet():561]: Current profile: {MY_PROFILE_DATA}
18:36:31.596 [RxCachedThreadScheduler-1520] D/PUMPQUEUE: [CommandQueue.isThisProfileSet():562]: New profile: {MY_PROFILE_DATA}
18:36:31.596 [RxCachedThreadScheduler-765] D/PUMPQUEUE: [CommandQueue.isThisProfileSet():562]: New profile: {MY_PROFILE_DATA}
18:36:31.596 [RxCachedThreadScheduler-1520] D/PUMPQUEUE: [CommandQueue.add():164]: Adding: CommandSetProfile - SET PROFILE
18:36:31.596 [RxCachedThreadScheduler-765] D/PUMPQUEUE: [CommandQueue.add():164]: Adding: CommandSetProfile - SET PROFILE
18:36:31.595 [RxCachedThreadScheduler-1519] D/PUMPQUEUE: [CommandQueue.isThisProfileSet():561]: Current profile: {MY_PROFILE_DATA}
18:36:31.597 [RxCachedThreadScheduler-1520] D/PUMPQUEUE: [CommandQueue.notifyAboutNewCommand():203]: Starting new thread
18:36:31.597 [RxCachedThreadScheduler-1519] D/PUMPQUEUE: [CommandQueue.isThisProfileSet():562]: New profile: {MY_PROFILE_DATA}
18:36:31.597 [RxCachedThreadScheduler-1519] D/PUMPQUEUE: [CommandQueue.add():164]: Adding: CommandSetProfile - SET PROFILE
18:36:31.597 [RxCachedThreadScheduler-1519] D/PUMPQUEUE: [CommandQueue.notifyAboutNewCommand():203]: Starting new thread
18:36:31.598 [RxCachedThreadScheduler-765] D/PUMPQUEUE: [CommandQueue.notifyAboutNewCommand():203]: Starting new thread
18:36:31.595 [RxCachedThreadScheduler-2609] D/PUMPQUEUE: [CommandQueue.isThisProfileSet():561]: Current profile: {MY_PROFILE_DATA}
18:36:31.598 [RxCachedThreadScheduler-2609] D/PUMPQUEUE: [CommandQueue.isThisProfileSet():562]: New profile: {MY_PROFILE_DATA}
18:36:31.598 [RxCachedThreadScheduler-2609] D/PUMPQUEUE: [CommandQueue.add():164]: Adding: CommandSetProfile - SET PROFILE
18:36:31.599 [RxCachedThreadScheduler-2609] D/PUMPQUEUE: [CommandQueue.notifyAboutNewCommand():203]: Starting new thread
18:36:31.604 [Thread-47887] D/PUMPQUEUE: [QueueThread.run():121]: connect
18:36:31.604 [Thread-47886] D/PUMPQUEUE: [QueueThread.run():121]: connect
18:36:31.604 [Thread-47885] D/PUMPQUEUE: [QueueThread.run():121]: connect
18:36:31.607 [Thread-47888] D/PUMPQUEUE: [QueueThread.run():114]: connecting 0
```

I see there 4x "Starting new thread". I checked also occurrences of "Waiting for previous thread finish" and "Thread is already running" but there are none around. So if we agree that there should be only one `QueueThread` running I see a chance of concurrency issues in `private var thread: QueueThread` field of `CommandQueue`. Maybe I'm completely wrong but for me it looks like candidate for `@Volatile` field. What do you think? Maybe I can try it during weekend (better than factory resetting the phone).

MilosKozak, 2021-04-23:
would be nice to more of logs prior this

vmichalec, 2021-04-23:
Ok, I did next round:

18:25 Profile switch -> 2x profile updated notification
18:28 Profile switch -> 2x profile updated notification
18:30 Profile switch -> 4x timeout error dialog, no success -> then at 18:35 3x profile updated notification
18:38 Profile switch -> 3x timeout error dialog, no success -> 18:42 1x timeout error dialog, no success
18:44 DanaR tab, clicking bluetooth icon to sync with the pump -> 1x profile updated notification (followed by some additional communication with pump, but finished)
18:47 Done

[20210423_ProfileSwitches.zip](https://github.com/nightscout/AndroidAPS/files/6366928/20210423_ProfileSwitches.zip)

vmichalec, 2021-04-23:
I have applied my change:

```
    @Volatile
    private var thread: QueueThread? = null
```

20:09 APK with updated CommandQueue installed
20:10 Profile switch -> 2x profile updated notification
20:13 Profile switch -> 2x profile updated notification
20:13 SMB
20:15 Profile switch -> 2x profile updated notification
20:16 Profile switch -> nothing happened until 20:18 -> SMB -> 2x profile updated notification
20:20 Profile switch -> 2x profile updated notification

Each time profile was updated successfully immediately (with 2x notification) or after some delay before pump communication. There were also some SMBs or other loop commands in between so there were more items in the queue sometimes (as expected). So for me it looks like it works. 

[20210423_ProfileSwitches_patched_with_volatile.zip](https://github.com/nightscout/AndroidAPS/files/6367579/20210423_ProfileSwitches_patched_with_volatile.zip)

vmichalec, 2021-05-09:
I have an update here. That `@Volalile` update is not fix of my problem. After few days of use it goes wrong again.

`CommandQueue` is designed as `@Singleton`. That's ok and reasonable. But there is also `independentConnect` method called exclusively from `DanaRExecutionService` (I'm on Dana R). This method creates new instances of `CommandQueue` for some purposes. After some invocations there are 2 or more instances of `CommandQueue`. Are those new instances only temporal? I don't know. There is `CommandQueue::init` method where is registration of each `CommandQueue` instance to `.toObservable(EventProfileNeedsUpdate::class.java)` (therefore long living instances?). This is also the case for those instances created by `independentConnect`. 

Here I think it is not intended to handle profile switches multiple times by multiple concurrent `CommandQueue`s.

I did some modification of logged messages with printing hashCode of instances:

```
// Application restarted by me followed by first CommandQueue instance with hashCode 150406160
13:04:58.265 [RxCachedThreadScheduler-85] D/PUMPBTCOMM: [SerialIOThread.disconnect():200]: Disconnected: Application exit
13:04:59.615 [main] D/PUMPQUEUE: [CommandQueue.<init>():109]: New CommandQueue instance created: 150406160, thread: Thread[main,5,main]

// Here is independentConnect which creates another CommandQueue instance with hashCode 267532805
15:16:25.856 [Thread-1100] D/PUMPQUEUE: [CommandQueue.independentConnect():215]: Starting new queue on Thread[Thread-1100,5,main]
15:16:25.857 [Thread-1100] D/PUMPQUEUE: [CommandQueue.<init>():109]: New CommandQueue instance created: 267532805, thread: Thread[Thread-1100,5,main]
15:16:25.858 [Thread-1100] D/PUMPQUEUE: [CommandQueue.independentConnect():217]: New queue on Thread[Thread-1100,5,main] started: 267532805
15:16:25.859 [Thread-1100] D/PUMPQUEUE: [CommandQueue.add():164]: Adding: CommandReadStatus - READSTATUS bolusingInterrupted 

// Some profile switch later on
16:37:39.467 [pool-26-thread-1] D/DATABASE: [DatabaseHelper$7PostRunnable.run():1807]: Firing EventProfileNeedsUpdate
16:37:39.468 [RxCachedThreadScheduler-98] D/DATATREATMENTS: [TreatmentsPlugin.initializeProfileSwitchData():232]: initializeProfileSwitchData
16:37:39.470 [RxCachedThreadScheduler-99] D/PROFILE: [CommandQueue$1.accept():114]: onProfileSwitch from Thread[RxCachedThreadScheduler-99,5,main] on instance 150406160 
16:37:39.470 [RxCachedThreadScheduler-230] D/PROFILE: [CommandQueue$1.accept():114]: onProfileSwitch from Thread[RxCachedThreadScheduler-230,5,main] on instance 267532805

// Another profile switch later on
18:53:49.033 [pool-26-thread-1] D/DATABASE: [DatabaseHelper$7PostRunnable.run():1807]: Firing EventProfileNeedsUpdate
18:53:49.035 [RxCachedThreadScheduler-99] D/PROFILE: [CommandQueue$1.accept():114]: onProfileSwitch from Thread[RxCachedThreadScheduler-99,5,main] on instance 150406160 
18:53:49.037 [RxCachedThreadScheduler-230] D/PROFILE: [CommandQueue$1.accept():114]: onProfileSwitch from Thread[RxCachedThreadScheduler-230,5,main] on instance 267532805
```

And here is a piece of older logs where things were really wrong with timeouts and error dialogs...
```
12:47:56.830 [RxCachedThreadScheduler-2633] D/PROFILE: [CommandQueue$1.accept():114]: onProfileSwitch
12:47:56.830 [RxCachedThreadScheduler-1581] D/PROFILE: [CommandQueue$1.accept():114]: onProfileSwitch
12:47:56.830 [RxCachedThreadScheduler-2300] D/PROFILE: [CommandQueue$1.accept():114]: onProfileSwitch
12:47:56.830 [RxCachedThreadScheduler-2299] D/PROFILE: [CommandQueue$1.accept():114]: onProfileSwitch
12:47:56.830 [RxCachedThreadScheduler-2632] D/PROFILE: [CommandQueue$1.accept():114]: onProfileSwitch
12:47:56.830 [RxCachedThreadScheduler-1578] D/PROFILE: [CommandQueue$1.accept():114]: onProfileSwitch
12:47:56.831 [RxCachedThreadScheduler-1970] D/PROFILE: [CommandQueue$1.accept():114]: onProfileSwitch
12:47:56.830 [RxCachedThreadScheduler-1579] D/PROFILE: [CommandQueue$1.accept():114]: onProfileSwitch
12:47:56.830 [RxCachedThreadScheduler-380] D/PROFILE: [CommandQueue$1.accept():114]: onProfileSwitch
12:47:56.836 [RxCachedThreadScheduler-1580] D/PROFILE: [CommandQueue$1.accept():114]: onProfileSwitch
12:47:56.837 [RxCachedThreadScheduler-1582] D/PROFILE: [CommandQueue$1.accept():114]: onProfileSwitch
12:47:56.837 [RxCachedThreadScheduler-1583] D/PROFILE: [CommandQueue$1.accept():114]: onProfileSwitch 
```

I will check how many concurrent instances I will have next days.

MilosKozak, 2021-05-11:
it can be the issue. tray call 
`tempCommandQueue.disposable.clear()` at the end of `independentConnect`
`independentConnect` is used when connection is broken during communication with R pump. singleton instance can hold BT connection in undefined state and simply call `readStatus()` don't work

MilosKozak, 2021-06-14:
https://github.com/nightscout/AndroidAPS/commit/fcf7faa3e05daf9c408821de1677ca1a6b6d7ab1

closed 2021-06-14 by MilosKozak

vmichalec, 2021-06-15:
Ok, from my point of view, this last fix solved the issues. Profile switching works as expected in few seconds without issues. Thanks