nightscout/AndroidAPS #485: Dexcom reporting high for over 45 minutes turns off loop

biwhite, 2021-04-18:
Dexcom high once reading 22.3 remains at that value.  If this continues for more than 45 minutes there is logic that kicks in to disable loop, to prevent a libre issue.

This looks to have been addressed when using the patched Dexcom app as a source, but not when using xDrip as a source.

Relevant previous issue is https://github.com/nightscout/AndroidAPS/issues/35

Can this same fix be put in when using the xDrip/Dexcom source please to prevent loop from turning off?

The relevant commit looks like it was https://github.com/nightscout/AndroidAPS/commit/3442e29e5c95a50edf85f838b922bb6294df7091

This  might just need an tweak to logic to check whether it's DexcomApp or xDrip with Dexcom source?

MilosKozak, 2021-04-18:
It's not planned at the moment. The reason is not technical but it's a question of trust what is real source of data.
xDrip is very complex and it's possible to fake the data. Thus only Dexcom app is accepted as trusted source now

closed 2021-04-18 by MilosKozak

biwhite, 2021-04-18:
Can you suggest a workaround for when we know this issue has been hit?  Entering a calibration reading to try and get it looping again doesn't work.  It seems the only way currently is to keep putting bolus in until the levels come down for readings to change?
Turning the loop off when BG is high, and we know is accurate is quite frustrating.

MilosKozak, 2021-04-20:
well if you are higher than 20 mmol and insuling is not working well i'd do manual action anyway. syringe???
you can use dexcom app along with xdrip

biwhite, 2021-04-20:
There was a pod failure which didn't scream when it ran out for some reason
so she ended up high.  First time that's happened.
We addressed it with some boluses, but didn't have confidence/understanding
as to why the loop wasn't working until I dug into all the old bug reports
looking for similar things.

As soon as the bolus eventually got the BG on the way down and new readings
appeared the loop was able to work again.

I guess it's just scary not knowing why it's not running, and having no way
in software to turn it back on or ignore the chunk of logic that's turned
it off for some time frame.
I'd be keen to understand why Dexcom app is trusted, but xDrip isn't?  My
girlfriend thought she'd seen some issues on various facebook groups with
the Dexcom patched app lately.  I'll have a dig into them and see if that's
the route we want to go.

Thanks