nightscout/AndroidAPS #492: Enter carbs  externally via nightscout

Simonsaccount, 2021-04-25:
Hi,

I wanted to ask, weather I did a mistake in my configuration or if the feature is missing:
I would like to be able entering carbs externally via nighscout - this would make it more easily for me to use the nighscout REST-API to add entries for example with my gramin smartwatch (no google wear).
However once I manually enter carbs in NS they do not appear in AAPS and thus are not considered in the loop.

Is there an option I missed to activate or is it intended that carbs can only be entered through the AAPS-Interface?

Best regards,
Simon

Tornado-Tim, 2021-04-27:
Nightscout Syncing via Pull is disabled in the latest version of AAPS as its a security risk. You can enable it still, but needs to explicit and involves several steps.
The Wiki has details.

Simonsaccount, 2021-04-27:
Hi Tim,
thanks for clarifying that it can work. I guess You are hinting towards this article, right?
[https://androidaps.readthedocs.io/en/latest/Installing-AndroidAPS/Nightscout.html?highlight=sync#androidaps-settings](https://androidaps.readthedocs.io/en/latest/Installing-AndroidAPS/Nightscout.html?highlight=sync#androidaps-settings)

I made sure that download from NS should be enabled. Under the Nightscout-Client Panel I confirm that the treatments seem to be received correctly in AAPS. They are however still not shown or considered in the algorithm.

Am I still missing a step?

MilosKozak, 2021-04-29:
do you have NS UPLOAD ONLY set to off?

Simonsaccount, 2021-05-02:
yes for sure.
Upload is working fine. And datalog shows that treatments are received correctly. I currently use V2.8.2.

[image]

MilosKozak, 2021-05-03:
you don't have enabled engineering mode and proper NSClient settings

Simonsaccount, 2021-05-03:
No I am currently on master-branch. So do I need to switch into Dev? Please elaborate.

MilosKozak, 2021-05-03:
https://androidaps.readthedocs.io/en/latest/Installing-AndroidAPS/Releasenotes.html?highlight=release%20notes#version-2-8-1-1

closed 2021-05-03 by MilosKozak

Simonsaccount, 2021-05-03:
Thank You. Works like a charm now :)