nightscout/AndroidAPS #496: No Support of Sony Smartwatch 3 SW50 since latest DEV (2.8.2) version

thomasmenges, 2021-04-28:
Hi I'm using the Sony smartwatch 3 SW50 in combination to AAPS. 

This works fine till version AAPS 2.8.2 master development, after implementing a newer google play service on the watch as it is described in
https://androidaps.readthedocs.io/en/latest/Usage/SonySW3.html

After creating 2.8.2 DEV in Android studio and installing it on the smartphone the watch isn't updated anymore.
So I tried to deploy the newest 2.8.2 watch.APK via ADB to the smartwatch without luck. The installing process cannot be done anymore. 

Using the 2.8.2 master version the installation of the watch.APK via ADB works fine.

Could someone please check, why the installation of the newest version of AAPS 2.8.2 on the SW50 isn't possible anymore ? Thanks

dlvoy, 2022-01-19:
@thomasmenges can you be more specific with errors you get while building AAPS wear target? Or when you try to install with `adb install` - you also should get either success confirmation or error message.

Since with current Android Studio (Arctic Fox 2020.3.1 Patch 4) and 3.0-rc3 i have no problems to build wear target - and to install it on Sony Smartwatch 3 (SWR50). Watch itself is correctly detected by ADB and can be debugged and monitored via Studio.

[image]

thomasmenges, 2022-01-20:
@dlvoy thanks for your information.
I'm sorry. I cannot check that  again,  because I switched to the new Samsung Watch 4.
Before that I stayed on the master 2.8.1 and that was ok on the SWR50. 

As you mentioned,  it seems that in RC3 the problems are solved.
So, in my opinion , this issue can be closed.

closed 2022-01-20 by MilosKozak