nightscout/AndroidAPS #500: Treatments older than 7 hours are IOB: 0.0 and not transmitted to xdrip+

amosxo, 2021-05-02:
The SMB symbols and treatments are no longer visible in xdrip+ since February the 20th 2021. They are visible thought in AAPS. I have not changed the settings in either of the apps. Also in the treatments log in AAPS I get IOB values from only the last 7 Hours. Before that period it only displays IOB: 0.0. I am using AAPS version 2.8.2 full release and 17cbcf0-2021.03.08 xdrip+ on Redmi note pro miui 12.0.2.0 android 10 QKQ1.191215.002. In xdrip+ interapp settings the local broadcast and accept treatments are enabled. I tried many options and I have rebuild and reinstalled AAPS and xdrip but it did not help.
Unfortunately the log before the issue occurred is now gone but I can provide the current log.

MilosKozak, 2021-05-03:
it's a question for xdrip group

MilosKozak, 2021-05-03:
treatmetns older than DIA are not contributing to IOB anymore

amosxo, 2021-05-04:
> treatmetns older than DIA are not contributing to IOB anymore

That makes sens.

closed 2021-05-05 by MilosKozak