nightscout/AndroidAPS #501: Rileylink/Omnipod Error 049

franzec, 2021-05-04:
DATETIME was 2021.05.02 09:15
Some minutes before the error occured, Aps lost connection to the Rileylink/Omnipod because of out-of-reachment.
Aps-build is 4588bccb9-2021.001.24-14:18
RL-fw is  BLE113: ble_rfspy 2.0, CC110: subg_rfspy 2.2
[nsmail.zip](https://github.com/nightscout/AndroidAPS/files/6420268/nsmail.zip)

--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

andyrozman, 2021-09-21:
This needs to be looked over by someone from Omnipod team, something is wrong with Omnipod packets.. 

Can you specify what hardware you have: RL (or clone, which one), what kind of phone, what OS.

franzec, 2021-09-26:
I had this error three times. My phone is Samsung S8 with Android 9.
Two errors with RileyLink and the last with OL (Firmware 2.6).

samspycher, 2021-10-05:
@andyrozman why do you think it is an error with Omnipod packets? At first glance, it looks like a standard screamer due to an actual delivery issue.

franzec, 2021-10-05:
Meantimes I lost another POD with this code. Samsung S8, actual Android9, AAPS 2.8.2, OL 2.6. I wear the OL In a leather-bag on my belt on the same side as the POD is. The POD-location is changing arond the belly.
There is another user with the same experience and environment.

05.10.2021 08:47:36 Sam Spycher ***@***.***>:

> @andyrozman why do you think it is an error with Omnipod packets? At first glance, it looks like a standard screamer due to an actual delivery issue.

franzec, 2021-10-15:
> @franzec[https://github.com/franzec] Do you have the exact same log entry as above? If you have a different issue, it would be super to receive a log file, and the time when it happened, in a new issue. If you aren't sure, feel free to add a file here, again with the time it happened. Thanks!

Hi Sam,
today, oct 15, 7:09 pm I got another error 049, about 2 hours after I set a new pod.
Environment is Samsung S8, Android 9, AAPS 2.8.2, OL 2.6.
The logfile is attached. Please tell me, if you need more Informations.
Greetings Franz

MilosKozak, 2021-11-15:
closed for inactivity

closed 2021-11-15 by MilosKozak