nightscout/AndroidAPS #504: [AAPS 2.8.2] Broadcasting Treatments into xDrip+ do not work without AAPS - NS connection.

reijoahola-nightscout, 2021-05-06:
This was tested with AndroidAPS 2.8.2 but probably the same problem is with earlier releases.
The latest xDrip+ Nightly Builds were used like 2021.05.05

When AAPS/NSClient is enabled, Local Broadcast enabled in AAPS/NSClient settings and xDrip+/Accept Treatments setting = ON, AAPS Treatments can be seen in xDrip+ main display beside CGM graph.

But when AAPS/NSClient -> NS connection fails or AAPS/NSClient is disabled then AAPS -> xDrip+ broadcast does not work.

A weird thing is that the broadcast even work when just NSClient is enabled but NS -> AAPS synching is disabled like it is by deafault in 2.8.2. It looks like AAPS -> xDrip+ broadcast use only data from NS and when it is not available, it does not work.

RalfCGM, 2021-05-06:
I think I have similar issues which might have the same cause. With my setup (aaps 2.8.2,xdrip latest nightly), there isn't any information coming from aaps directly to xdrip, no matter what settings I enable or disable in aaps or xdrip. I think this happened since I erased all aaps data trom the phone and reloaded a stored prefs file. Maybe theres a hidden setting stored in the prefs file regarding the connection between aaps and xdrip? The direct  connection from xdrip to aaps works flawlessly.

muthijammy, 2021-05-06:
I had enabled NS upload only with wifi in xdrip+(dont usecmobile data nightscout upload). As a result, when on mobile data, I see similar issue of broadcast not working

closed 2021-10-08 by MilosKozak