nightscout/AndroidAPS #532: Autosense calculates duplicated entries

hartlijn, 2021-05-23:
2.8.2 (bug was present in v2.3 and v2.6 as reported here https://github.com/MilosKozak/AndroidAPS/issues/1897)

Autosense is taking dupes into account for a long period.

Bg source is xDrip with G6.

[image]
[image]

[AndroidAPS._2021-05-23_00-00-00_.0.zip](https://github.com/nightscout/AndroidAPS/files/6527723/AndroidAPS._2021-05-23_00-00-00_.0.zip)
[AndroidAPS._2021-05-23_00-00-00_.1.zip](https://github.com/nightscout/AndroidAPS/files/6527724/AndroidAPS._2021-05-23_00-00-00_.1.zip)
[AndroidAPS._2021-05-23_00-00-00_.2.zip](https://github.com/nightscout/AndroidAPS/files/6527725/AndroidAPS._2021-05-23_00-00-00_.2.zip)
[AndroidAPS._2021-05-23_00-00-00_.3.zip](https://github.com/nightscout/AndroidAPS/files/6527726/AndroidAPS._2021-05-23_00-00-00_.3.zip)
[AndroidAPS._2021-05-23_00-00-00_.4.zip](https://github.com/nightscout/AndroidAPS/files/6527727/AndroidAPS._2021-05-23_00-00-00_.4.zip)
[AndroidAPS._2021-05-23_00-00-00_.5.zip](https://github.com/nightscout/AndroidAPS/files/6527728/AndroidAPS._2021-05-23_00-00-00_.5.zip)
[AndroidAPS._2021-05-23_09-20-50_.0.zip](https://github.com/nightscout/AndroidAPS/files/6527729/AndroidAPS._2021-05-23_09-20-50_.0.zip)

hartlijn, 2021-05-24:
Just happened today again. 15 mins ago all was fine. Autosense on 1.0 and no duplicates. I check now and autosense is on 1,85 / 1,15 with all duplicate values.

Makes me uncertain to use Autosense any longer

MilosKozak, 2021-06-04:
this issue is rare and AAPS recover from it itself usually. So far I was not able to locate the issue because it's rare and not permanent. I found out it's related to some irregularity in bg data

joelm335, 2021-07-02:
I have the exact same issue.  I do not find that AAPS recovers from this by itself - a forced restart of the phone is required. The issue then usually returns within a few hours.  My BG source is G6 plus xDrip

MilosKozak, 2021-11-10:
restest with 3.0 and create new issue if needed

closed 2021-11-10 by MilosKozak