nightscout/AndroidAPS #539: Bolus delivery error while being delivered. Not recorded in NS.

bchris21, 2021-05-26:
During the last week I have experienced 3 times the following:
1. I give bolus using the wizard.
2. Bolus delivery error appears on AAPS
3. Bolus is being delivered when I manually check the pump.
4. Bolus not recorded in NS, unless I click "Refresh" button in Medtronic Menu and get the pump's recent history.
5. Carbs also not recorded in NS. I have to manually add them in AAPS.

It would be pretty dangerous for my 3y old son for getting a double bolus if I haven't checked the pump.

-Latest (2.8.2) AAPS used. 
-NS version 14.2.2
-Orangelink (Latest Firmware update)
-Medtronic 554

In which email can I share the logs? Thanks

andyrozman, 2021-06-04:
Please share the logs (here). But f there was delivery bolus error, then this wouldn't be recorded in NS. So we need to see what is happening... 
Pump history is read every 5 minutes, so treatment would be recorded in NS in max 5 minutes later (but probably before). 

Does this happen every time? So far nobody said anything that bolus delivery doesn't work on 554 and driver has been in use for about 2 years now.

Alamo04, 2021-06-05:
Had that also, but only.... perhaps 2 times in the last 2 month.

After next history reading the bolus was shown in NS, but not the carbs. Thats normal, because AAPS didn't make entrys.
Carbs are not stored in pump and can't be read with the history.

Alamo04, 2021-06-06:
@andyrozman 
happens just now. Log sendet and attached.
[AndroidAPS_LOG_1623004128483.log.zip](https://github.com/nightscout/AndroidAPS/files/6604667/AndroidAPS_LOG_1623004128483.log.zip)

bchris21, 2021-06-21:
Sorry for the late reply. As @Alamo04 said, it also happened 2-3 times for the same period. I have decided to change pump and I am now in AC Combo having 554 as spare.

closed 2022-05-21 by bchris21