nightscout/AndroidAPS #546: Pump not reachable. Watchdog sleeping

egrehm, 2021-05-30:
### The Setup for our child is:
* AAPS 2.8.2
* Dexcom g6 (patched app)
* Omnipod Eros
* Emalink (nano&micro)  ble_rfspy3.13 / subg_rfspy 2.2.21
* Galaxy s7 with lineageos 17.1

### The Problem:
AAPS looses connection to the pump and Watchdog can not fix it. Propably because it is shown as connected in the Omnipod tab.

### Questions:
1. Is there any Option to extend the Watchdog to toggle bluetooth on the Pump not reachable event even if is shown connected (this fixes the issues in far most of the issues)?
2. Would it be possible to add SMS command like 'toggle bluetooth' in case above is not an option?

andyrozman, 2021-06-04:
Answers:
1. Definetly not.
2. No. Watchdog actually does toggle, only problem is that toggle needs to be longer that < 1s... But maybe we could extend that... or make it configurable... 
Since you are using Lineage "toggle BT" should work correctly on your system...

egrehm, 2021-06-04:
From the manual bt toggle i would say that 1 sek is enough but did not stop the time. 

> Since you are using Lineage "toggle BT" should work correctly on your system...

even if it is not intended for my rig?

> 1. Definetly not.
Is watchdog for omnipod/riley planned? If not maybe the config text should show the supported setups?

egrehm, 2021-06-18:
We have to toggle bluetooth multiple times a day (and night :-(  ). It would be so valuable to have a self healing function. In best case not waiting the 30 min until reacting. 
Is there an option for official aaps  watchdog support for emalink/onmipod?
Is there a usable workaround?  I have no good enough and simple idea yet.

Foxy7, 2024-02-11:
Bumping this one, as my wife's phone has repeatedly had the same issue for both Samsung S22 and now S23.

I mentioned this back in Aug23 on discord but no one else commented on it.

[image]

The last day it happened was the 10th Jan, where my wife lost connection all night.

[image]

I had a look in the code and realised that 1 second is too short a time frame for the watchdog bark. so increased this to 10s for all three steps.

in this file:
<kotlin/app/aaps/implementation/queue/QueueThread.kt>

```ruby
                        pump.disconnect("watchdog")
                        SystemClock.sleep(10000)
                        (context.getSystemService(Context.BLUETOOTH_SERVICE) as BluetoothManager?)?.adapter?.let { bluetoothAdapter ->
                            bluetoothAdapter.safeDisable(10000)
                            bluetoothAdapter.safeEnable(10000)
```

I changed lines 81, 83 and 84 to have 10000 instead of 1000 and the issue appears to be solved.

I also noticed that a quick manual bluetooth toggle failed to fix the issue, whereas a longer one did, which is why i choose 10s.

MilosKozak, 2024-02-12:
It looks your phone specific.
But anyway .... once we  increase target API one more time, BT will not be accessible from app and this code will be removed at all
My recommendation is change your phone

Foxy7, 2024-03-28:
understood, thanks.

we can close this one.

closed 2024-03-28 by MilosKozak