nightscout/AndroidAPS #551: [dev-e5] graphs stopped to be updated

Philoul, 2021-06-06:
Sometime secondary graphs update is stopped see screenshots below: (I had this issue twice, last time this morning)
- 1st screenshot: this morning when I saw the issue
- Basal curve and all secondary graphs are "frozen" (only BG value and SMB remains updated
- If I change time range (see 2nd screenshot), curves remain frozen (no update of vertical scale, if time range is longer no update of curve for previous values, only horizontal scale is updated...)
- loop continue to works and NS is still updated
- if I restart AAPS, graphs are immediatly updated

[image]

I can share my log if it can help...

MilosKozak, 2021-06-09:
it will be necessary to connect debugger when this happen and look at cached data

Philoul, 2021-06-21:
I catched the bug (see extract with log added at beginning and end of each preparexxx function)
=> See synthesis below and complete log if usefull informations

Bug at 13:08:23.245
Just before it entered in prepareIobAutosensData() but I don't get log of end line
`13:08:20.781 [RxCachedThreadScheduler-10] D/UI: [OverviewData.prepareIobAutosensData():614]: prepareIobAutosensData()`

after bug no more `D/UI: [OverviewPlugin.refreshLoop():247]: refreshLoop finished`
Only `prepareBucketedData()` and `prepareBgData()` is executed

```
13:07:48.075 [RxCachedThreadScheduler-51] D/UI: [OverviewData.prepareBucketedData():398]: prepareBucketedData()	
13:07:48.076 [RxCachedThreadScheduler-51] D/UI: [OverviewData.prepareBucketedData():411]: prepareBucketedData()  from 1624240900000 to 1624276900000	
13:07:48.077 [RxCachedThreadScheduler-51] D/UI: [OverviewData.prepareBgData():336]: prepareBgData()	
13:07:48.120 [RxCachedThreadScheduler-51] D/UI: [OverviewData.prepareBgData():350]: prepareBgData()  from 1624240900000 to 1624276900000	
13:07:50.290 [OverviewFragmentHandler] D/UI: [OverviewData.preparePredictions():356]: preparePredictions()	
13:07:50.293 [OverviewFragmentHandler] D/UI: [OverviewData.preparePredictions():392]: preparePredictions() from 1624240900000 to 1624276900000	
13:07:50.294 [OverviewFragmentHandler] D/UI: [OverviewData.prepareBasalData():417]: prepareBasalData()	
13:08:02.679 [OverviewFragmentHandler] D/UI: [OverviewData.prepareBasalData():510]: prepareBasalData()  from 1624240900000 to 1624276900000	
13:08:02.680 [OverviewFragmentHandler] D/UI: [OverviewData.prepareTemporaryTargetData():516]: prepareTemporaryTargetData()	
13:08:02.976 [OverviewFragmentHandler] D/UI: [OverviewData.prepareTemporaryTargetData():547]: prepareTemporaryTargetData() from 1624240900000 to 1624287524860	
13:08:02.977 [OverviewFragmentHandler] D/UI: [OverviewData.prepareTreatmentsData():553]: prepareTreatmentsData()	
13:08:02.994 [OverviewFragmentHandler] D/UI: [OverviewData.prepareTreatmentsData():608]: prepareTreatmentsData() from 1624240900000 to 1624276900000	
13:08:02.995 [OverviewFragmentHandler] D/UI: [OverviewData.prepareIobAutosensData():614]: prepareIobAutosensData()	
13:08:15.934 [OverviewFragmentHandler] D/UI: [OverviewData.prepareIobAutosensData():836]: prepareIobAutosensData() from 1624240900000 to 1624276900000	
13:08:15.935 [OverviewFragmentHandler] D/UI: [OverviewPlugin.refreshLoop():247]: refreshLoop finished	
13:08:15.936 [RxCachedThreadScheduler-51] D/UI: [OverviewData.prepareBucketedData():398]: prepareBucketedData()	
13:08:15.937 [RxCachedThreadScheduler-51] D/UI: [OverviewData.prepareBucketedData():411]: prepareBucketedData()  from 1624240900000 to 1624276900000	
13:08:15.938 [RxCachedThreadScheduler-51] D/UI: [OverviewData.prepareBgData():336]: prepareBgData()	
13:08:15.981 [RxCachedThreadScheduler-51] D/UI: [OverviewData.prepareBgData():350]: prepareBgData()  from 1624240900000 to 1624276900000	
13:08:16.991 [RxCachedThreadScheduler-10] D/UI: [OverviewData.preparePredictions():356]: preparePredictions()	
13:08:16.993 [RxCachedThreadScheduler-10] D/UI: [OverviewData.preparePredictions():392]: preparePredictions() from 1624240900000 to 1624276900000	
13:08:16.994 [RxCachedThreadScheduler-10] D/UI: [OverviewData.prepareBasalData():417]: prepareBasalData()	
13:08:20.367 [RxCachedThreadScheduler-10] D/UI: [OverviewData.prepareBasalData():510]: prepareBasalData()  from 1624240900000 to 1624276900000	
13:08:20.369 [RxCachedThreadScheduler-10] D/UI: [OverviewData.prepareTemporaryTargetData():516]: prepareTemporaryTargetData()	
13:08:20.753 [RxCachedThreadScheduler-10] D/UI: [OverviewData.prepareTemporaryTargetData():547]: prepareTemporaryTargetData() from 1624240900000 to 1624287799651	
13:08:20.754 [RxCachedThreadScheduler-10] D/UI: [OverviewData.prepareTreatmentsData():553]: prepareTreatmentsData()	
13:08:20.780 [RxCachedThreadScheduler-10] D/UI: [OverviewData.prepareTreatmentsData():608]: prepareTreatmentsData() from 1624240900000 to 1624276900000	
13:08:20.781 [RxCachedThreadScheduler-10] D/UI: [OverviewData.prepareIobAutosensData():614]: prepareIobAutosensData()	
13:08:23.185 [RxCachedThreadScheduler-63] D/UI: [OverviewData.preparePredictions():356]: preparePredictions()	
13:08:23.188 [RxCachedThreadScheduler-63] D/UI: [OverviewData.preparePredictions():392]: preparePredictions() from 1624240900000 to 1624276900000	
13:08:23.245 [RxCachedThreadScheduler-10] E/CORE: [FabricPrivacy.logException():81]: Exception: 	
java.lang.IllegalArgumentException: Failed requirement.	
	at info.nightscout.androidaps.database.interfaces.DBEntryWithTimeAndDurationKt.setEnd(DBEntryWithTimeAndDuration.kt:11) ~[na:0.0]
	at info.nightscout.androidaps.plugins.iob.iobCobCalculator.IobCobCalculatorPlugin.getCalculationToTimeTempBasals(IobCobCalculatorPlugin.kt:589) ~[na:0.0]
	at info.nightscout.androidaps.plugins.iob.iobCobCalculator.IobCobCalculatorPlugin.calculateFromTreatmentsAndTemps(IobCobCalculatorPlugin.kt:206) ~[na:0.0]
	at info.nightscout.androidaps.plugins.iob.iobCobCalculator.IobCobCalculatorPlugin.calculateIobArrayForSMB(IobCobCalculatorPlugin.kt:335) ~[na:0.0]
	at info.nightscout.androidaps.plugins.general.overview.OverviewData.prepareIobAutosensData(OverviewData.kt:756) ~[na:0.0]
	at info.nightscout.androidaps.plugins.general.overview.OverviewPlugin.refreshLoop(OverviewPlugin.kt:245) ~[na:0.0]
[... cut ...]
13:17:57.326 [RxCachedThreadScheduler-51] D/UI: [OverviewData.prepareBucketedData():411]: prepareBucketedData()  from 1624240900000 to 1624276900000	
13:17:57.327 [RxCachedThreadScheduler-51] D/UI: [OverviewData.prepareBgData():336]: prepareBgData()	
13:17:57.376 [RxCachedThreadScheduler-51] D/UI: [OverviewData.prepareBgData():350]: prepareBgData()  from 1624240900000 to 1624276900000	

```

[AndroidAPS._2021-06-21_11-26-26_.2.zip](https://github.com/nightscout/AndroidAPS/files/6688687/AndroidAPS._2021-06-21_11-26-26_.2.zip)

Philoul, 2021-06-22:
I propose to close this issue. I'll reopen it later if the fix doesn't work...

closed 2021-06-22 by Philoul