nightscout/AndroidAPS #552: [dev-e5] NS some careportal events are updated in NS and removed or lost a few hours later

Philoul, 2021-06-06:
Issue with Insulin Age, Sensor Age and Site Change (I don't know if it's the same for other CP events...)
When I update this information in AAPS, it's updated in NS, but a few hours later (I didn't understand exactly what is the trigger, but I think the day after), these informations are "lost in NS (but remains Ok in AAPS)

1rst screenshot the 4th of june in the evening (14 hours after insuline and Site change) information Ok in NS (note it's the same for sensor change but last change was 3 days earlier so KO in this screenshot)
2nd screenshot the 5th of june in the morning, information is removed
[image]

Philoul, 2021-06-11:
Yesterday same issue (I changed Insulin and site in the morning and sensor in the afternoon)
=> 10 hours after Insulin and site change (and immediatly after sensor change), everything was correctly updated in NS, but about 2 hours later, information lost in NS...

[image]

Unfortunatly no log available (bug in latest dev)...

swissalpine, 2021-06-11:
Seems related to https://github.com/nightscout/AndroidAPS/issues/456

Philoul, 2021-06-12:
@swissalpine 👍 it's the same issue...

closed 2021-06-12 by Philoul