nightscout/AndroidAPS #569: [dev e5]  Dana-i  - error reports data coming from different pump

MichaelJohnSmith, 2021-06-19:
Using latest code from dev branch at 18Jun2021
[image]

Problem is intermittent/random?  In log files below, first instance of error at 16:17:35.310
If looks as if the pump sometimes returns DanaR rather than DanaRS, or the string is being truncated.

16:17:35.310 [RxCachedThreadScheduler-1] D/NOTIFICATION: [NotificationStore.add():60]: Notification received: Data is coming from different pump. Change pump driver to reset pump state.
16:17:35.311 [Binder:10606_4] E/PUMP: [PumpSyncImplementation.confirmActivePump():73]: Ignoring pump history record  Allowed: 18/06/2021 07:40:20 DanaR VBJ00081FI Received: 1624025039000 18/06/2021 15:03:59 DanaRS VBJ00081FI

Error seems to result in loss of returned data.  Commands execute ok but responses may be lost.  Occassionally, failure results in need to re-pair the pump to AAPS
Pump was received at the end of May.  
Country: GBR  Build date: 25/Jan 2021  Version: FPN-1.0.1
AAPS reports:
Firmware: Dana-i (BLE5) 
Model: 09
Protocol: 12
Code: 12

[AndroidAPS._2021-06-18_21-36-28_.log](https://github.com/nightscout/AndroidAPS/files/6681377/AndroidAPS._2021-06-18_21-36-28_.log)
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/6681355/AndroidAPS.log)

I have managed to "work around" the failures by modifying the confirmActivePump function in PumpSyncImplementation.kt

**replacing:**
if (type.description == storedType && serialNumber == storedSerial && timestamp >= storedTimestamp) {
**by:**
if (serialNumber == storedSerial) {
            if(type.description != storedType) {
                aapsLogger.warn(LTag.PUMP, "MJS - ignoring pump type change - Expected: $storedType Received: ${type.description}")
            }

This code has now been running successfully for 24 hours, in open-loop mode, executing boluses and temporary basals and monitoring pump status with no visible errors - though the log file confirms that the errors still occur but are ignored.

MilosKozak, 2021-06-21:
fixed by https://github.com/nightscout/AndroidAPS/commit/d0365afe899d37ede3db1d931693edf4db09d76d
you need switch to virtual pump and back in order to proper initialization

MilosKozak, 2021-06-21:
thanks for catching this bug

closed 2021-06-21 by MilosKozak

MichaelJohnSmith, 2021-06-21:
Thanks for the quick fix.  Let me know if you need any testing effort.