nightscout/AndroidAPS #570: 2.8.2 Random Crashes

m2karam, 2021-06-20:
Hello Team,
I've been using AAPS for few weeks and noticed it crashes almost once every 2-3 days, sometimes it crashes overnight or when I'm not using the phone, however a couple of times it crashed right after delivering a bolus. 
In the attached logs you'll see it crashed around 8:00am and I only noticed later around 9:40.
Please let me know if you need any further details, thanks in advance!
[Log Export.zip](https://github.com/nightscout/AndroidAPS/files/6682451/Log.Export.zip)

MilosKozak, 2021-06-21:
it's exactly at 8AM.
99% PlayProtect. Disable it

m2karam, 2021-06-22:
> 
> 
> it's exactly at 8AM.
> 99% PlayProtect. Disable it

Thanks Milos! I disabled it today and will keep an eye on it for the next few days and let you know how it goes.

closed 2021-06-26 by m2karam