nightscout/AndroidAPS #581: [2.8.2.2-Dev] Temporary basal does not reset to 0% when pump is disconnected

fjpezuela, 2021-06-25:
Reporting bugs
--------------

When disconnecting the pump, the temporary basal does not reset to 0% in the main AAPS screen.  The baseline value for that hour is displayed.  However, if you change the screen, you can see that the temporary basal is at 0%.
[image]
[image]

Version: 2.8.2.2-Dev Build 33047e478

fjpezuela, 2021-06-26:
Add that I have disconnected the pump more times these days and it has not happened again. This may occur in certain circumstances that I am not aware of.

MilosKozak, 2021-06-27:
ok, download logs if you see this issue again and open new issue

closed 2021-06-27 by MilosKozak