nightscout/AndroidAPS #582: [2.8.2-Dev] Insulin boluses and carbohydrates not syncing with xDrip +

fjpezuela, 2021-06-25:
Reporting bugs
--------------
When you add a treatment from the calculator, the data is synced correctly in Nightscout, however it is not synced with xDrip +.  No carbohydrate or bolus treatment is displayed in xDrip +.
[image]
[image]
[image]

Version: 2.8.2.2-Dev Build 33047e478

fjpezuela, 2021-06-25:
SMBs are also not displayed in xDrip +.  All this with version 2.8.2 worked without any problem

MilosKozak, 2021-06-27:
hopefully fixed by https://github.com/nightscout/AndroidAPS/commit/272cdbbb0010a1acd9043de3629ce8131ef2d44b
remember it's still broadcasting only NSClient data ie only what is comming from NS

hartlijn, 2021-07-02:
not fixed unfortunately (in todays dev version). 
xDrip always received treatments (smb, bolus, carb and even notes if I'not mistaken) directly from AAPS.

not sure what you @MilosKozak mean with 'only what is coming **from** NS'. I have both xDrip and AAPS configured to only uploadto  (and not dowload from) NS. To be able to have my system running not require an internet connection.

fjpezuela, 2021-07-02:
It doesn't work for me either after recompiling the app

MilosKozak, 2021-10-08:
sorry i was wrong. there was never direct AAPS -> xDrip broadcast. up to version 2.8.2 NSClient did broadcast data comming from NS. With new NSClient this functionality was removed

closed 2021-10-08 by MilosKozak