nightscout/AndroidAPS #584: [2.8.2.2-Dev] Unable to delete IOB

fjpezuela, 2021-06-25:
Reporting bugs
--------------
After giving a treatment, when you go to the treatments option, the "Remove" link does not appear to be able to remove the active insulin.  The link if active to remove carbs.

[image]

Version: 2.8.2.2-Dev Build 33047e478

andyrozman, 2021-06-25:
Can you please start learning how to use software and stop opening invalid tickets.

fjpezuela, 2021-06-26:
I know how to use the software. With version 2.8.2 I can delete the IOB without problem, the delete button appears. In this version it lets me remove COB, notes, basals, but I don't see the option to remove the IOB as in 2.8.2.

fjpezuela, 2021-06-26:
I'm just trying to help. I have been reporting the things that I have seen that do not work or do not behave the same as it does with version 2.8.2.

I'm sorry if I've caused a nuisance.

MilosKozak, 2021-06-27:
IOB translation in your language is too long. Change it in Crowdin
You may test the feature in English language

closed 2021-06-27 by MilosKozak