nightscout/AndroidAPS #585: [2.8.2.2-Dev] Some texts and images are shown in English with language in Spanish

fjpezuela, 2021-06-25:
Reporting bugs
--------------
Using the Spanish language, some application options display texts and images in English.  For example, the buttons for time goals are displayed in English.  Many texts of the treatment part are shown in English.  A choice of the graphics is displayed in English.

[image]
[image]
[image]
[image]

Version: 2.8.2.2-Dev Build 33047e478

Alamo04, 2021-06-26:
This is not a master version.
Dev versions are still "under construction" and most of your reported issues are known by the developers.
So, if you don't find a realy bad issue, don't post it here.

After the dev, it comes a Release Candidate.
There you can look, search and test the version for issues and report everything you find.

fjpezuela, 2021-06-26:
Hello @Alamo04 ! Yes, I agree with everything you say. I only created it by the post on Facebook that we created the tickets with all the problems we detected. It's not really a problem as such, but I decided to comment on it more than anything because I guess the developers will test the English versions more, and they may not be aware of this.

For me there is no problem in closing this ticket.

MilosKozak, 2021-06-27:
Spanish translation is not finished. Let's do it on Crowdin ;)

closed 2021-06-27 by MilosKozak