nightscout/AndroidAPS #587: [2.8.2] SMS - 2 boluses delivered during 1 min. - limit ignored

xmizi, 2021-06-28:
**AndroidAPS 2.8.2 (build info in attached screenshot), engeneering mode enabled**
_Note: it was not a test, just a coincidence_ 

Limit set to 15 min, 2 allowed phone number (XXXXXXXXX and YYYYYYYYY)
1. bolus (number XXXXXXXXX): 1U
2. bolus (number YYYYYYYYY): 0.5U

both buluses was delivered. 
_____________________________________________________________
Filtered log:

08:54:27.070 [IntentService[DataService]] D/SMS: [SmsCommunicatorPlugin.processSms():206]: SMS from +XXXXXXXXX: Bolus 1
08:54:27.091 [IntentService[DataService]] D/SMS: [SmsCommunicatorPlugin.sendSMS():982]: Sending SMS to +XXXXXXXXX: K potvrzeni bolusu 1,00U odpovez SMS s kodem z aplikace Authenticator pro: Lukin nasledovano kodem PIN
08:54:42.925 [IntentService[DataService]] D/SMS: [SmsCommunicatorPlugin.processSms():206]: SMS from +XXXXXXXX: 9360227643
08:54:42.930 [IntentService[DataService]] D/SMS: [AuthRequest.action():55]: Processing confirmed SMS: Bolus 1

08:55:02.679 [IntentService[DataService]] D/SMS: [SmsCommunicatorPlugin.processSms():206]: SMS from +YYYYYYYYY: Bolus 0.5
08:55:02.682 [IntentService[DataService]] D/SMS: [SmsCommunicatorPlugin.sendSMS():982]: Sending SMS to +YYYYYYYYY: K potvrzeni bolusu 0,50U odpovez SMS s kodem z aplikace Authenticator pro: Lukin nasledovano kodem PIN
08:55:16.855 [IntentService[DataService]] D/SMS: [SmsCommunicatorPlugin.processSms():206]: SMS from +YYYYYYYYY: 8258337643
08:55:16.858 [IntentService[DataService]] D/SMS: [AuthRequest.action():55]: Processing confirmed SMS: Bolus 0.5

08:55:17.209 [Thread-19215] D/SMS: [SmsCommunicatorPlugin.sendSMS():982]: Sending SMS to +YYYYYYYYY: Bolus 1,00 U aplikovan uspesne
08:55:17.243 [Thread-19215] D/SMS: [SmsCommunicatorPlugin.sendSMS():982]: Sending SMS to +XXXXXXXXX: Bolus 1,00 U aplikovan uspesne
08:55:23.594 [Thread-19215] D/SMS: [SmsCommunicatorPlugin.sendSMS():982]: Sending SMS to +420721747071: Bolus 0,50 U aplikovan uspesne
08:55:23.631 [Thread-19215] D/SMS: [SmsCommunicatorPlugin.sendSMS():982]: Sending SMS to +XXXXXXXXX: Bolus 0,50 U aplikovan uspesne
_____________________________________________________________

**Build info**
[image]

**incident log**
[SMS_problem.txt](https://github.com/nightscout/AndroidAPS/files/6725406/SMS_problem.txt)

MilosKozak, 2021-06-28:
this should be already resolved in dev. can you test it too?

xmizi, 2021-07-07:
Thanks for reply Miloš. We will be careful next time. It was only one random case.
2.8.2 works without any problem. If not necesarry, I'll no install dev version.

closed 2021-07-07 by MilosKozak