nightscout/AndroidAPS #597: [2.8.2.2-dev] smb delayed

hartlijn, 2021-07-03:
To be honest i am not sure how this worked on master release. It felt like SMBs were not given as fast as i expected. Pump= dana RS

SMB takes a long time after requested time to be given.
[image]

margogoodfellow, 2021-07-06:
Looks like you have the same problem as me. Your loop didn't refresh with bg change. Mine does the same sometimes

MilosKozak, 2021-07-06:
Do you have log?

margogoodfellow, 2021-07-06:
Hey yes i do. I added link to it. Itd stored on my Google drive if you
click on it i will authorise access other wise i dont know how to send it?

> Do you have log?

PK  ���R               AndroidAPS.log��[�%�u-���+
:~� u9��np��� Ye��0�d�j��-t7������j^*3W���1�k�%�Ad������+��>
�#k��4����������L���~������������������?�����������?t���W?��Gn4�}t��o^�z����/�=|����Wo_�{������Oo^��]�����v��������>�M��������{����_?���/�~�h������1�]�36v�O.�������^���\��������_��W�������<|����]w��__<�����u�u�u?���ya3��~�p����Y�5�/n�6���0�|>|����o?:�����=]������O��?�������=�����7_?YeU��Y����Z���{����o�������������������0���nn�������|����~���O>2��������/�����������Ow~<]�_�����T����'9�n{��9����|{�x����Xm�N?Wv�������������<��P�����7�^�W5�>S��.>~vw������?��������~��������?�����������?���z���?����_~���������������/�y���{�������������������y�����������'_{����}�'}q����������w��gOYtG4�|G����_<�}�5����_|��q��w������~������=��?��}�������zM�����1�\��{v���=����o�����Z_����5�����?����U�^||Z��g���^������~d��������~��7�������7���V�������o����mWV���������=�2^���U���-�j���lU�����JQ�{�v��&��RT+�F~V���n����U�5����x���3g����RrQ�X��O���v���mQm��OE�w�zz���FN�E�RQ��'u�����8wS�*�
~�6�bac���*��/�rY�����Vw?�Ba��}���Ra}�����nT,���e�P��C����X��oWc��^w~|������~t����lm�����RYC��uZVu��iM���nZ]���GV�9]MbM�?�W�nY�~�'�����������z�t��������zz��8	���kS�������N�+��j�,B�pEGW���#�?�������z�\7�]�8Qq��u�����n��n�B9���=��j���G
0��~s������	���)C������~����p��W;�����|G,+�1����E)��t�gg��&�����c�If|&q�h�N��IS�|6q��H�d��� �s�s��D�g��9���$�s����� �l����R"���E_���Q~4�oV���5��m��Y��l��pu�~a���u��u����>z>Dn�P����W��d��WtU��~�J����x��l����p������(�����a���k�
�;��?$C>�9�p��L�<`�M"��������3�A)�Of���e��2�c�"�3���1g����y��� �t�<`�Y�|6rA�,b>9���,���b,���(�FS�v���6p�,�q����F�[wc�ni�#���(����)��|���}�),:�����U,{W����z>f����~x��@~����.�����d���������l�<)a>�9OZ��e���3�3����3�����3�I%�Oh�e/���|:r����\��;�\>]��m&�F3�v���]������#����V�|��n�B����&�\6�2�������R&?�����r��=�5�a�!�T%/��#?� �wu���!����QzK>�9{���|.s���M@��2g���|"s���M@�'2gL��|6s�e��
(���}�,�|�g3g_6�� ����tZY6�S�3j��LtQ6��.��w�D������u�`;>��~A>l��t��t���|[����)!?>��.!�/\�UA~��3��h������]������Q�6�|�����`����!������y3h$���\��&2>�7�������e��r������y�S�|:o��B>�7�������C������	�@��b������M��U?>��j��B������m���u7��/��7I��VU�^"��gE�?%�_6h�"�l�a���1v��w�z�o�����A8����5�����������sYsP"|&kZ~�d�:sn��Yg�-�5���%�f�I�t��3�������RQ���[*jVsKE�G��h
��R�}#�o4���e���pK�n2��r������	��SR�<#n}�����_��*�������QV�c�o��_y��M|.i�q�$�Zn��TT����44X|6i	�N�Ac�e���f��`���47X�{&i���
�AS���	�Q#�E����q�8�����*?6��Fcm��^�~�����m�B9|��K�{�4��O����^��N��>g������l��'7�{7��vY���Y�}�l�d�7�|�����������A����sf��`����4(X|:g�
����u���9��X[*g�jK��Zm��Ym�-3�N���"4e�������F�]���`v����7w��t{Z�6p��W�^�Q�!>8��!!{/\�5�}�5�����=�����x7�,�e�����n���t�j"�w�xO�4mU<]rAs M[����h��(�l�@CWe���fTYE�g���5��	���$[&hZSl��9hM��rf���T��6���+����}XF�C�����<W�����Z�=uK�FxM����^�����	��P:����l�j����1v���3F��a�'1���u�Z����uS���A���ZBh��_K�:l��Yg�-4k���f��T��5��
��F�RA���Z*gV\K��j�k��Bp���f X����K�o4�64��\�k��6K�n2�����������oY��WtUz_:�������Jv�cK��3��q�����Z7�����Z7��=3���%df���t��3�������R1���Z*f��\K��ZSk��Ykf-3k���Rf���T�\sf-�V���]�z��_}ui��f�"����F3k��������q��}���������3k��9d��_�U�}�x^��#�lst�y�xv��V�)����p .+�1��z����!Y����t^p��s!3h�h�T�*�,�L��,�=2���r����I���S!3h
�,�L��,�=2�f �r�d���g����QC�e�����1�0�7���Y�������e�fYv����[����q����n/��qsr������#t���Kg�Jv�����]{�}$~V��"z�����@�����M�M��>]����tXp��S!3h�,�L�,�=2�F�r����1���S!3h�,�L� ,�=2� �r����������A�����Q3�E�g���k�����Y&������]�F�{����~����2�^q{M���eS�>!y��G�^��oU����=i����'�����������>���/�>�l�!�����/W\ �D����_��{&dFU�{"d�A#�E�gB�4�Z�{"d�A��E�gBfTQe�'B�4�����"sS-�=�1��*�=�1�F�'�=�U�M����&��2y��A��l��Y��\o��u��w��a����r�^o{���jv�[6��mB���E��i��[��>��S��xR=���b'RC| z��k��O?,�>]�������rO�����@�gB��Xx��S!3����3!���CE��Bf����r�������{^d��0�,�\��0�{�����a��	z�e����vOe�|PrM��U���y�;r�,�n�_T|:w��^�/i�P��_����i������]a_���t���G���ac�},�>m�No��KOo��>H���{��{W7t����J��R��J��R��J����r��TT�����+��������*3��2�s)�W���)�W���!3��2�S!����sa�o��w���[��}��=p�,��R����/,����a����r�^o{�7K��}@�h)��]J�>~EWe����,�}�o���p�������f���Ptr;&�v-�����(�6]���9`���S!s�de�gB���1�rO��=��$�=2��v�9��E������{%��C�^K����W�{*c5A����=[�t�d}�����{�
�� {h��u�����q)�X��������~P����}���
��Zp�� �g<������=�l�~XV�cvo�v?��{�l�T�\Z�l�gb�sh����LCD�{&f0�sv���#�!"�=3������t�<b�"��s1��i��zO��#�%"�=�2�Zx������t��c�|C���h���b�c��=p�,�n������a�nx{Z�6n����^���O��A�����}���J���F���g ����)cN>�������1��4N�e��VV�c~���~����.����y2o���������L�<a�"�����'LSD�{.j�0mQ���y�0�����y�0�Q���y�0m����y�0]���U������M�M,����g���_&��.��R���.z��f��u����g�[w��v�����B���=nQ��`4����pE�d���4����?��B���q�����{�?���
y����2��T5|;%>7F����y0����QE�����i���O��F�����*)>7��*>7[LgDD|&n����L�\�l��N�������m��!���,�`9~|,}=��{���f	���r�b�����>Hp�����F�_���nUr�3���"�gv��q�~h+��vOF<��s?���Z�]�A2Q��t�g����C����I�@�����x����s�����������O������S�3h����L��.#>8�����O���a�	�����y�����E��e��G��a���������u�\��~9>l�t�%t��q|�Ddq{m6y9��S���+�*���I>�����nyC�}+��?��Ztso��A~��O��������&0';�s|^r.]��\�\6:_���TX�����t�x��S�3hR���T�.B>9�&����E��Y�����3h^�l�l��.[>�8���'H>� [... cut ...]
�I�mJ5��J��d�R�6%CcU MjlS���D4�6��M�B����� �����)Y�"�Qc�Ru.���yy,j�R��=u)Y""lQc�}��bf`bI�����&�w���Yx7����v{1�k��������`�Eu��J��sWy�x�T�(�����(Gv����S9(���2�.q���{�z�c��5j�]�>6�E^�l.5��"�����1��O]au����
Q����
�������mt+*Dy/t�Bl2�`�
���a��6mcS�S�Z�T�E65����N�
�J��]j�T���D4����S�B���?���N��J��\�>5v*U�T"�����T�S�h�#�Sc�R�N%����O��JU:�<�=��M��J�9���=�.56*m���2*�D����F���?5*������_Q��g}��]jrr����b��>]��������`E����m>w�ok������f�W�(�%��>����k������]�"��;�R�#��^;��h�V��4[k{vF&�����V[���� 3Y�D�(TF���J�"���^���$%cT�\���h��9�j��q��������U%��5Z)�8�C�X�Rz�\|N��j1F��xi|Jp�&��6��Rt~������!Gl	7mb�

hartlijn, 2021-07-06:
Log attached. See timestamp as in screenshot above.
[AndroidAPS._2021-07-05_00-00-04_.52.zip](https://github.com/nightscout/AndroidAPS/files/6773096/AndroidAPS._2021-07-05_00-00-04_.52.zip)

MilosKozak, 2021-07-07:
Fixed by https://github.com/nightscout/AndroidAPS/commit/ad98bcec2eca55c25f06eb0dd123ce33081f38a9
Interresting issue related to time synchronization. 
I thought only RS with protocol >= 6 can set time with seconds but it looks like your RS with protocol == 5 is capable of this too

closed 2021-07-07 by MilosKozak

hartlijn, 2021-07-09:
confirmed solved.

I've always had this issue that AAPS was waiting for a while (up to a minute) before connection was established, resulting in the annoying overlay 'waiting for' message as described in this issue: https://github.com/MilosKozak/AndroidAPS/issues/2668 which seems to be solved now. 

That issue seems to be solved now as well. Nice job, @MilosKozak