nightscout/AndroidAPS #598: APS doesn't not respond to BG readings

margogoodfellow, 2021-07-04:
Reporting bugs
--------------it happening all day today. BG is changing every 5 min but APS doesn't run sometimes for 25 min i have to press run now so it will act accordingly to the BG. I have deleted and  installed it again but it didn't help. I had this issue before. But then it fixed itself work fine for a bit and again it does not respond. Im attaching logs and it will be aroud or just after 20:00. 

> ``
[https://drive.google.com/file/d/1HACT5HAy2kbuqoDMmFu79XAxkBuGEE4W/view?usp=drivesdk]()

MilosKozak, 2021-07-07:
asked for access to file

margogoodfellow, 2021-07-07:
Hi. I gave you access so should be ok now

MilosKozak, 2021-07-07:
What version of AAPS is it?
Log contains only 15 min of data
turn off AUTOSENS logging
Provide long from longer timeframe

margogoodfellow, 2021-07-07:
Its 2.8.2 i have lots of logs with this issue on my drive. Shall i share
all of them? I deleted the app and installed it again and it working ok
now. But this was 3rd time this happened and i had to delete everything and
install it again
  Log Export
<https://drive.google.com/file/d/10VMDfNC8xdkz0y1srWGAhnncId4TbDNl/view?usp=drivesdk>

  Log Export
<https://drive.google.com/file/d/123bzMe0m4dJI8HqsoZNoqRRqtcm-J03i/view?usp=drivesdk>

PK  ���R               AndroidAPS.log��ms����������0���L,7�$�kE��C���U86���1��H=$G;���ts��K"���8�����2DU���]YWg���V����"�������r�VR�u���|��?~��������O��n���~wW���.�>��z�N	��w�n>6�������_~��������m�P��H�a�����iw���o_=\�z���������~(/��!6�������y�6��W�o�E��?��U�pu{s)�O�?��Yl�������_���lv�����w�?}{����Z��w���?2�)��}���]��������?���[qa��]�<O\H�)w�<�o������O�����/�4�m�~�Z1���Q�������~�����}���M�|~W�\^�\���i���f�7*��d�~��c�����O���Fx�	_O���������o�k�|e����#����Z�<��h	_��������I��/�������<��|��rDx��w�:W�0�rA"S��d����L�'���d���I4��d�Bd���Oq��(�$��^��Nc��o'}Q��� �=�d�Nn�v2�N2�N�F'9��~��D�~z��������'�<��u2��-�jO�S�$��=yaR�Oz�N:��N��i~i��j�+�9�k?�M}���mO=������c3��w�	�6���_����_���z�u��_n�����Z�<\\���������-m3���~��o��{���������u��/������uV�F4�������/7W��V^_�/7w�/7��������/O�����������j����������}�������{���Q>|����W���T?��������/����w�?���/��?X��5=X}w+X����h9XP��Q���h_e�r�����E���Z.����}������C���kb��������'�Y�������+������j����m���b�Y�	�������n�<��W������o��]���������������������M��������7����OR���������{}��������o������w����?)�������!}�k���,^��j��z�����N�������R��<��O�����~�U�d_���`���#��� ����n?�����������\��������}����]�hy�}�Q�2X�2Pf�:0c��j{�pW���8������U}��z�Y����6����_��U=���bs}�s���}�F^��X���?���4���<�s"e�q�����5�~w����������;Hv�>Dw�w����	~�����~y���M��g5m\k#�������u�v��u�$Uj�8�&��/���T�����>����e�������'��#'-��ea4��=�":X��1����9�S ~&��12���o'"�N���������_�n�����]y��p}�������6�7�������C=����7����Z�F���� 
13	���V���S5�zKR^�_�w���e}���e}��Z���m3����;��f����t+�e:Y��W&���-�:�Q�Y��Oq����_Dyg,�,��m�ig��3���������(��D�,��g$|3���4�l�	����������=��t������5|`���of�~Vf�'�F���-Hj����:����v����0�y��3��}K����d����Cj�/|RC5�v�S���Hv��>#��������	o��Z��6����Cy�f-.|k�Y��i(��� �h�I�c6+L&�G��
�I�����Y��	G�8m=_�YyA2~�~��sR���B�w����N!��A�td��G�
�����5j*	p2-�.��8�Bp��n�?W�Q�N�N�$��(
 t�>�u0��+���>�����^Z�LO�O�W����H�N@����3BW 2�0���!t��e���;���w����e����]�f�s�����lD'���/��F�cB��Fd���d#�9����F��$��]�N�<T�����T�b��9
�1K�2�J����T�b�4�
�1K�2�J���8D�,��HM���4�4
�1K���������q��Y� �h�fyc�*L&G���
���B��"�������1J5_$����q�9@�Y�:�m�^�l��+L/��p��	V���t��g�QGI��a��z�x��uW1a�w#�n�s*�s��	t2�V3�F7
^+Y3���a�G��i`V&����������?D*:x��� ov���\��2����u�V�+�B8���K�u�kos�
����D7�|��������#������v�1s�&�����������sO+��,�� �1��.���8��sO+��,��o�=-�������tD�7���mA3���{
���KD�4���d��q<��� �h(�!<��'B�h(BOc�i�F��4��DJ��q2x�H��ij����bs�[M�b\:��3�V�8�����ad�-��^wF�4I*#I#���(��i�>��0�Ns���:m���|��<��T�@��)������<z�:��91c���{-];��`-�,�YD#�x 0{*���uX������W�
�&(��H^���I�	�T	b�T�e���l�*��e��1��e��1�S�e�-M�����e�1Z\����
q����
%�1Z�L8	��Q�L4\�	���L4\%N�1b4_#����0
�������1a4_$���8s$���6Z�Lu�����M�������7:�6���o78X`�F***@***.***�����
�b
hi����i( �b
hq�;�D���
�I%*�P�L��v1&S��]L��lS@T%���P�F�
v1���%�
�)U�.�����l�.=Z�K�U���t�g��D:�s��`q!�3���I,0XTG� �d���^��`-���b�(�Ev������|D��&~S�Zt������0�2�I�������o�q-�V�BG���^��F�k9;�\W��f%�q`k����rp�$�#��b��=K�G�2��r]^k��o0����Z��,Mo���y�c����Y\������3x-�v�d�P����0�h@��bl'H&�`��bl�P&� �����n��U��hDS�G������UL���D�F{\��N�N4�Z����h�L3O;����JX������t�����������B�n9XP���s��5����@�X
�Hb��uW1a��=K�hj����D�!�������@U0�F0^��2�5����*GV�o
^5������U@��:�vK���Y��9 [��zY���r�g���ig�dP*u,�`-�VH��oO����������Y�l2k����������Y+Fn�d��P2���0�h ��b�&H&N`��b��P&T '����*�bQ��hDS�����Q�M�b�&H%����	��hS��QL��D���8����c
".���2��SE: ���>�!pz����g�;�,�F} &��:�>����;B^�C �u{(�a9�����&��������0�����:�c��K�PfB�)�tg��(1�t�=���_����
��4��@�?+*l+�S��4��8W J���)J�d;���3�I';qb�%d1�m�x.���K�L4$1�$B1$1�B(***@***.***$&]�kD�P���t��C��HL��DS<���t	���@�$]�T"���<�.*5+Dq$N6��4kGK�#��i2�2MR��F^>a��A#�A��uW|��M����A����A:�S2�>G�T��gO���g�Hd��(����'P��M�Ht������0Ho�$9+�����:N��&�jBF� O��J3n����:��H��~�$�9)��D]+�`�.R�[\(>I�\���2�d2]����CM$��d�����D�^�$#2A2��C$#2�2��8����l��w�#2��8����|�h�W�#2A*��8���L�L4�J21"�|��LL��D��E9&&dBT���N��8����w�qLv���m�U�Y:����9Y�r��`���9,�`Q���XO>��d����8&���b�(���8�.��1�������D����s]3I�����$SQ����D��s
�d�ZQ�i8fL2���Q���B=��=��#[e��9e�O���n����r��/�hY ***@***.***�������8����|�h��cbF&@#�������}�rL���D_�$32A2��(���L�J4�N21# }��L����D���A&Fd��D��8����7�R�L��l� ***@***.***;fs ����B��9c6V�6�/>XJP��\O1����.~)@&���b��T/�X��cp?FE� �����(��
eB��2��U��_�I�5��2M(����j�G���1�4���c��� s����2ue����f%�'�L��@��O�C�C$�>M_�%�LQ�P&�A�������2�1����$i�ebFf�F4�GP��� �h��cbFf�F4��(���L�J4��8���L�L4�(���L�J���� ���	��VBA&fdBD�adp����"4��21"�Q��8�����F��L���� S����4�L�	���R,0X2�,���������s����2���Cx�UL�����Vr�V&�����,�������F��b��b&�d"M��&��a:�y�aQ��)L�-�����K��h%3�����(b��w���F�*�	��`A%���o��d�V��-i"�$��J���^ID����!c0��vqT�1�����AQ%�`�T���pX�1� ��]b&H�Du��	)Q�%�`BDJTt�1��]b
[... cut ...] ~�� �� �q�AX{lFe����W4���-X{g��':(?��Gq��z��^�������8���n~W��Gq������&|�����6s��[��&����4��������gc ����:kO4�&���GeVY{M���@@���#���O*�l��OUH>�'�v��l`�{�B��|�'g6��;�~p7�;b1,�����.�ui���!���@��X��0�d6/\���u�&�Ax����u���������R[5�T���s�Wt���R6�AG����7�����OW|m����v|l/Jm��}���FJy[�l����-H}����tR���
��h?��s���W�r~?]9�r��m��Q��m���h<66h��94�� �Gq����4�m<S}n4�YM�V��6@�Q����i��h�h<j3W������6@�Q�U4^����Mm���2�h� ��m���2kh� ���m���UG�M�'������</���#

margogoodfellow, 2021-07-08:
https://drive.google.com/file/d/1L0Nk2P8B_NrzeOQ3WbcofBJJbCTYSInH/view?usp=drivesdk

That's from just now. I have made this link available without permission

MilosKozak, 2021-07-08:
AUTOSENS logging is still on. it means log is full only by this and is really short

margogoodfellow, 2021-07-08:
Ok i switched this off and as soon as this will happen again i will send
the logs

margogoodfellow, 2021-07-23:
https://drive.google.com/file/d/1Th-YUShPICnE1By0NT7a7cw_-KPMaxW5/view?usp=drivesdk

It didn't work aroud 13:20 for 15 min. I had to press run now

margogoodfellow, 2021-07-27:
https://drive.google.com/file/d/1Y6NyOzKWtGK93C706dQmvqFQrz_3WLuK/view?usp=drivesdk 

It happened again at 18:40.
Over 10 min no response from aps. I have switched autotune so hooe log is ok

margogoodfellow, 2021-07-28:
https://drive.google.com/file/d/1Y6NyOzKWtGK93C706dQmvqFQrz_3WLuK/view?usp=drivesdk

[image]

MilosKozak, 2021-09-21:
AUTOSENS logging is still ON
log doesn't contain the right period

mick711, 2021-10-15:
[image]

mick711, 2021-10-15:
[image]
[image]
For me I can see that on openAPS glycemi show correctly but not on AAPS 2.8.2.5dev build 79f0fc

margogoodfellow, 2021-10-16:
https://drive.google.com/file/d/1UJWREu0WNm5X6Rf0__SVxTKwBYPfjpZk/view?usp=drivesdk

No response between 17:20 -17:35 had to press run now

Autosene is off in the log setting

MilosKozak, 2021-10-17:
I see execution 17:23:46
and then new BG @17:33:37
so your problem is no BG data. It can be sensor out of range or sleeping Dexcom app

closed 2021-11-10 by MilosKozak