nightscout/AndroidAPS #600: 2.8.2 master / Arrow direction bug

robertrub, 2021-07-05:
I've remarked this problem often. All the numbers are going up but the arrow _in the app_ is horizontal. What is new is that the arrow in the AAPS **watchface** is correct and going up...
[image]
[image]
[image]

MilosKozak, 2021-07-05:
on main screen AAPS uses direction provided by BG source. On watch may be calculated

andyrozman, 2021-07-05:
Arrow is correct on Xdrip and wrong in AAPS.

MilosKozak, 2021-07-06:
what source?

andyrozman, 2021-07-06:
XDrip+. It seems that for small changes arrow is always pointing from left to right ->, in Xdrip there is arrow that is just little slanted, I think in AAPS that one doesn't exists...

robertrub, 2021-07-06:
Look at my screen captures (all done during the same minute). G6, xDrip+ and AAPS 2.8.2. xDrip and AAPS _watchface_ show a 45° arrow (with delta of +7) but AAPS phone shows an horizontal one.

woernsn, 2021-07-06:
> 
> 
> XDrip+. It seems that for small changes arrow is always pointing from left to right ->, in Xdrip there is arrow that is just little slanted, I think in AAPS that one doesn't exists...

What I see in the code, the `FortyFiveUp` arrow should be shown with delta +7 in 5 minutes.
In that case, the `slope_by_minute` should result in 1.4.
```
if (slope_by_minute <= (-3.5)) {
    arrow = "DoubleDown";
} else if (slope_by_minute <= (-2)) {
    arrow = "SingleDown";
} else if (slope_by_minute <= (-1)) {
    arrow = "FortyFiveDown";
} else if (slope_by_minute <= (1)) {
    arrow = "Flat";
} else if (slope_by_minute <= (2)) {
    arrow = "FortyFiveUp";
} else if (slope_by_minute <= (3.5)) {
    arrow = "SingleUp";
} else if (slope_by_minute <= (40)) {
    arrow = "DoubleUp";
}
```

MilosKozak, 2021-07-07:
this calculation is used only if arrow is not provided by bgsource

StephenBrown2, 2021-07-08:
Check that backfill from Nightscout is turned off. If it's on, AAPS gets a duplicate number and thus the delta for the arrow is zero.

robertrub, 2021-07-08:
Hi Stephen. Yes D/L from NS is ON but if this was the problem, I'd always get a "double the normal slope" arrow (delta +7 AND delta +7) And, the watchface of AAPS is showing the correct arrow direction. It seems (but haven't been able to test it seriously), it is as if AAPS arrow on phone is (very) delayed compared to the actual status. 
Il my memory is correct, I had seen this when I was using the patched Dexcom app too.

MilosKozak, 2021-07-08:
Dexcom arrow shows longer change, not only difference between last 2 BGs

robertrub, 2021-07-10:
Another impressive screen captures. 
[image]
I have a quick look at AAPS (phone) and I say, "things are OK" by looking at the arrow.
[image]
If I look at xDrip and AAPS watchface, they both show a dangerous down arrow...

Philoul, 2021-07-10:
I noticed that sometime G6 updates previous version of his history values
for ex G6 can send at 8:00 a current bg value of 130
and 5 min later new value for 8:00 of 135 and for 8:05  a current value of 138
G6 arrow is then flat: 135 -> 138 < 5 mg/dl
In all app you get an up arrow: 130 -> 138 > 5 mg/dl

robertrub, 2021-07-10:
Hi ;)
What I can't understand is wht AAPS **watchface** and xDrip show _one and same_ type of slope and AAPS **phone* shows another....

Philoul, 2021-07-10:
AAPS shows arrow sent by source, watchfaxe and other apps show calculated arrows
=> With my example AAPS will show the flat arrow sent by G6 APP (even if calculated short delta is +8 mg/dl) while all other app will calculate a 45° up arrow

robertrub, 2021-07-10:
Well, info on AAPS vs other sources not being the same is confusing. AAPS phone vs AAPS watchface not being the same is VERY confusing (for me, definitely a bug).
In my last capture (BG rising, peaking and going down) I definitely prefer a down arrow at the beginning of the drop than in 10+ (?) minutes when the BG "average" will start to go down in AAPS.

robertrub, 2021-07-10:
[image]