nightscout/AndroidAPS #601: [2.8.2.2 dev] Direct bG read from xDrip+ not working but NSClient read of data from xDrip+ works fine

MichaelJohnSmith, 2021-07-06:
[image]

In the log file attached:
   
AAPS displays bG readings at 5 min intervals until
**21:43** Changed BG Source from NSClient to xDrip+
No new bG readings are displayed until
**22:26** Changed BG Source from xDrip+ to NSClient 
and bG readings resume

[AndroidAPS2.log](https://github.com/nightscout/AndroidAPS/files/6773087/AndroidAPS2.log)

xDrip+ version 17cbcf0-2021.03.08  - current release
using Libre2 (both direct and via MM2) + OOP2 
Dana-i pump
Samsung Galaxy A20e Android 11 (also same with Android 10 before upgrade)

MilosKozak, 2021-07-08:
you don't have proper setup in xDrip (package name). Check wiki how to setup xdrip

MichaelJohnSmith, 2021-07-08:
Thank you.  Setting Identify Receiver sorted the problem.  The problem started soon after changing from 2.8.2 Release to the development code.  Obviously too quick to blame the last thing changed!

closed 2021-07-08 by MichaelJohnSmith