nightscout/AndroidAPS #611: [2.8.2.2 - dev]  Target value not shown correctly on Home screen

MichaelJohnSmith, 2021-07-20:
I modified my local profile, changing both the high and low targets from 6.0 to 7.5, then saved and activated the profile at about 17:21.
I discovered the Home screen still showed a target of 6.0, both on the Profile button and on the graph.
[image]
I checked that the profile had be correctly saved.
The loop screen shows that the correct target, 7.5, is being used in calculations.
[image]

[image]
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/6850037/AndroidAPS.log)

Libre2 - OOP2 - Xdrip+
Dana-i pump
Samsung Galaxy A20e running Android 11
Just started Objective 6

MichaelJohnSmith, 2021-07-23:
On setting a temporary target, the screen is updated as expected.  On ending the TT, the screen shows to the same, wrong, target value.

After a phone restart (and the consequent restart of AAPS), the correct target value is displayed.

So it seems that an App restart provides a work-around for this issue.

MilosKozak, 2021-09-20:
resolved in dev

closed 2021-09-20 by MilosKozak