nightscout/AndroidAPS #618: Feature request : Automation / If TT = _value_ then ...

robertrub, 2021-07-29:
It vould be very useful if we had the possibility to test for a value for temp. target in automations. Actually, the only option is if a TT exists or not.

Usage : if TT set to 140 (exercice) then select sport profile (or use actual profile at 50%) fir x min.

TIA

robertrub, 2021-07-31:
Another use : I have automations to slow down steep positive deltas (TT 75 for 5 min) but I don't want them to kick in when I set TT to 140 and eat fast carbs before exercice... 

If TT <>140 and BG >130 and delta >6 then set TT= 75 for 5 min

SPL34, 2021-07-31:
Hi, I will also be very interested in the possibility of specifying the TT to better manage periods of sport or low risk.  It would also be a good improvement for automatisms to be able to trigger an action when a specific profile exists.  IF Profile x, then ... For the moment, we cannot use the different profiles as conditions and that is missing.

PR #690 (2021-10-03): Add new Automation trigger Temp Target value
On a new Trigger on the Value of a Temp Target (Equal, Greater , Lesser, Not Available...)

resolve #618