nightscout/AndroidAPS #619: NSClient Full Sync => "Infinite duration" (> 1 month) / and new BG values from G6 not updated in NS during sync

Philoul, 2021-07-30:
v 2.8.2.2-dev

I think it's a real bug of full sync function : I decided about 40 hours ago to click on Full Sync in NSClient...

At the beginning it was quite fast (about 1 value/s but still remaining #67500 BG values, #1200 TempTarget, #1950 TherpyEvents, #11400 boluses, #1050 carbs, #24800 temp Basal, #3800 Extended bolus
=> Raw estimation of 31 hours for the full synchronization... Quite a long duration!

But 40 hours later, it still remains 25400 BG values, but it takes now about 2 min per value (each value is sent and acked #30 ot 40 times...)
=> new estimated time :  1.5 or 2 months to finish the full sync !!!

And during the sync no new BG value received from Dexcom patched app is updated in NS web site...

I don't know how to stop the synchronization (when I stop and restart AAPS, synchronisation continue, I also tried Clear Log, Restart and Deliver now buttons but nothing works...)
[image]

Philoul, 2021-07-30:
After a new stop and restart of AAPS it started again to sync at a better rate...
=> I restarted AAPS because it becames very very slow and didn't receive several new BG values from Dexcom patched app...
=> Generally this version of AAPS restart "alone" about once a day, but if it doesn't it becomes very slow... So it's better to restart it manually...

[image]

Philoul, 2021-07-30:
And every about 2 hours, the number of sent / acked seems to be multiplied by 2 for each record...
[image]

Philoul, 2021-08-04:
finally, the full sync duration was about 50 hours (during this time no BG values were uploaded to NS), and I had to restart AAPS 3-4 times. (without restart I think the duration would have been much longer ...)

MilosKozak, 2021-08-11:
I agree it's slowing down during time. Unfortunately I don't know what's wrong ..... Maybe memory management in Android or java
ad double sent entries:
this code should prevent it `        if (startId == lastGvId && dateUtil.now() - lastGvTime < 5000) return false`
no idea why it don't work

MilosKozak, 2021-08-11:
i made sync variable `@Volatile` now

MilosKozak, 2021-10-17:
resolved by https://github.com/nightscout/AndroidAPS/commit/516630215c3adfb32577a27bea2d9a09be58b03b

closed 2021-10-17 by MilosKozak