nightscout/AndroidAPS #621: [2.8.2.2-dev] Statistic activity too long to get result

Philoul, 2021-08-04:
When I click in 3 dots menu Statistic :
- After 5s black screen
- After 20s a pop up "AAPS is not responding Close App / Wait" => Here it's necessary to select select "wait"
- screen remains black for more thant 1 min
- Results appears about 1min45s after selection of statistic in menu

I had to use it several times before understanding that AAPS was not crashed but was just to long to show the results :-(

Alamo04, 2021-08-04:
Confirmed that it takes a lot longer to get the results.
If you have a slow Smartphone, it can be as described above, with a faster one it takes 5-10 seconds to get something back.

Philoul, 2021-08-05:
My loop phone is not a "very slow phone" (Samsung A40), but it starts to have a lot of data in database
=> It's a bit quicker on my dev phone (a slow low cost phone) that have less data in its database because I often switch to different versions of AAPS (current master, current dev, several other branches...).
I noticed that Profile Helper also starts to take more time than previsouly...

MilosKozak, 2021-08-11:
fixed by https://github.com/nightscout/AndroidAPS/commit/c5857da42ff0ffa18dd46a7acdf317df1031a09f

closed 2021-08-11 by MilosKozak