nightscout/AndroidAPS #628: 2.8.2.2 dev 06/08/2021 and old

overfrenk, 2021-08-09:
Salutations. I am testing all the devs and even with the last one it often happens to me that the loop breaks without errors. Case # 435. And I always have to reinstall the old modified version "Increasing the fallback threshold to 200". It is really sad when it stops during the night and wakes up with blood sugar at 300! Would it be possible to have this setting available in the settings? Possibly only in engineering mode? Unfortunately eversense in some smartphones or configurations from these troubles due to esel !!!! 
When this problem occurs, I realize it because: smb no longer delivers anything since in LOOP it indicates no aps selected and in home under sensitivity it shows NeN.
Thank you.

MilosKozak, 2021-08-11:
i don't know what is "Increasing the fallback threshold to 200" version
you may try provide logs with data causing AAPS fail

overfrenk, 2021-08-11:
Hi,the problem is this : https://github.com/nightscout/AndroidAPS/issues/435

[image]

overfrenk, 2021-08-12:
yes, with default setting I often have the problem, while with 200 never!

overfrenk, 2021-08-22:
Almost 1 hour ago it happened again! I have tried various aaps and smartphone restarts. Nothing does not recover! I send logs via aaps ...

[image]

Autosense is empty!

[AndroidAPS_LOG_1629661695154.log.zip](https://github.com/nightscout/AndroidAPS/files/7028049/AndroidAPS_LOG_1629661695154.log.zip)

overfrenk, 2021-08-22:
19:56:29.487 [RxCachedThreadScheduler-36] D/AUTOSENS: [AutosensDataStore.getLastAutosensData():136]: AUTOSENSDATA null: data is old (broadcast) size()=779 lastData=22/08/21 19:40:00
19:56:29.512 [RxCachedThreadScheduler-23] D/AUTOSENS: [AutosensDataStore.getLastAutosensData():136]: AUTOSENSDATA null: data is old (PersistentNotificationPlugin) size()=779 lastData=22/08/21 19:40:00
19:56:29.514 [RxCachedThreadScheduler-23] D/AUTOSENS: [AutosensDataStore.getLastAutosensData():136]: AUTOSENSDATA null: data is old (PersistentNotificationPlugin) size()=779 lastData=22/08/21 19:40:00
19:56:29.600 [RxCachedThreadScheduler-36] D/AUTOSENS: [AutosensDataStore.getLastAutosensData():136]: AUTOSENSDATA null: data is old (broadcast) size()=779 lastData=22/08/21 19:40:00
19:56:29.638 [RxCachedThreadScheduler-23] D/AUTOSENS: [AutosensDataStore.getLastAutosensData():136]: AUTOSENSDATA null: data is old (PersistentNotificationPlugin) size()=779 lastData=22/08/21 19:40:00
19:56:29.642 [RxCachedThreadScheduler-21] D/AUTOSENS: [AutosensDataStore.getLastAutosensData():136]: AUTOSENSDATA null: data is old (PersistentNotificationPlugin) size()=779 lastData=22/08/21 19:40:00
19:56:29.643 [RxCachedThreadScheduler-23] D/AUTOSENS: [AutosensDataStore.getLastAutosensData():136]: AUTOSENSDATA null: data is old (PersistentNotificationPlugin) size()=779 lastData=22/08/21 19:40:00
19:56:29.644 [RxCachedThreadScheduler-21] D/AUTOSENS: [AutosensDataStore.getLastAutosensData():136]: AUTOSENSDATA null: data is old (PersistentNotificationPlugin) size()=779 lastData=22/08/21 19:40:00
19:56:29.689 [Thread-34767] D/AUTOSENS: [IobCobOref1Thread.run():97]: Prev data time: 21/08/21 10:10
19:56:29.696 [Thread-34767] D/AUTOSENS: [IobCobOref1Thread.run():105]: Aborting calculation thread (trigger): onEventNewHistoryData
19:56:29.698 [Thread-34767] D/AUTOSENS: [IobCobOref1Thread.run():329]: AUTOSENSDATA thread ended: onEventNewHistoryData
19:56:29.698 [Thread-34767] D/AUTOSENS: [Profiler.log():13]: >>> IobCobOref1Thread <<< executed in 222 miliseconds
19:56:29.763 [RxCachedThreadScheduler-9] D/AUTOSENS: [IobCobCalculatorPlugin.stopCalculation():361]: Calculation thread stopped: onEventNewHistoryData
19:56:29.764 [RxCachedThreadScheduler-9] D/AUTOSENS: [IobCobCalculatorPlugin.newHistoryData():383]: Invalidating cached data to: 22/08/21 19:51:28
19:56:29.765 [RxCachedThreadScheduler-9] D/AUTOSENS: [IobCobCalculatorPlugin.runCalculation():366]: Starting calculation thread: onEventNewHistoryData to 22/08/21 19:56:29
19:56:29.806 [Thread-34773] D/AUTOSENS: [IobCobOref1Thread.run():77]: AUTOSENSDATA thread started: onEventNewHistoryData
19:56:29.817 [Thread-34773] D/AUTOSENS: [IobCobOref1Thread.run():97]: Prev data time: 21/08/21 10:10
19:56:29.824 [Thread-34773] D/AUTOSENS: [IobCobOref1Thread.run():122]: Processing calculation thread: onEventNewHistoryData (2/408)
I think this is the mistake!

overfrenk, 2021-09-13:
Hello @MilosKozak . Do you think this problem of mine can be solved? Even today I had to restore my modified MASTER version because the dev broke down!

MilosKozak, 2021-09-13:
can you give me access to your BGs (NS address)? Read only is enough. If yes PM me
I cannot replicate your issue

overfrenk, 2021-09-13:
@MilosKozak Of course, there are no problems. Even if the problem occurs once in a week or several times ........ It has happened to me even once in a month !!! How can I get your PM? I can not find it.

MilosKozak, 2021-09-13:
https://www.facebook.com/milos.kozak/

MilosKozak, 2021-10-03:
fixed by https://github.com/nightscout/AndroidAPS/commit/f790c06b27235a8be929ef04cc007d13bf08b1c7

closed 2021-10-03 by MilosKozak