nightscout/AndroidAPS #629: BT connection lost when screen turns off and AAPS is not on the front screen

Technixx, 2021-08-11:
Preconditions:
- AAPS 2.8.2
- Nightscout 14.0
- Phone: Unihertz Jelly 2
- All battery optimiztations is turned off
- All required requests has been granted

When this happens: consistently and reprodusible
What is the circumstances: AAPS is on not "on top/front screen", click the "turn off screen" physical button
What happen: AAPS looses the BT connection to Orange/Rileylink, and the AAPS app is closed. When the screen is turned on again, you pass through the lock screen the AAPS app needs full start up, connect to BT and get Nightscout data.

Apps this does not happen with: Xdrip+
App can ble in the "background" but still have the BT connection open when the physical "turn off screen" is pressed

I have added a video of producing the behaviour. I have to add logs etc when I have access to the phone.

Video of behaviour: https://youtu.be/BMEsYWf_RHE

swissalpine, 2021-08-11:
Unfortunately, this is a problem of the Jelly 2's energy-saving measures.
I actually have this problem well under control when I set the cleaning time to 5 minutes under "App Restriction" (Smart Help) or completely disable the "App Restriction" function.

Technixx, 2021-08-11:
I will double check the app restrictions, but I also see that XDrip+ is functioning even when it is in the background and the screen is off. Maybe there are workarounds or lessons learned from xdrip+?

Technixx, 2021-08-11:
.

closed 2021-08-11 by Technixx