nightscout/AndroidAPS #630: Only to be sure that the Dev's be aware of some issues. All are in dev versions with the new database until today.

Alamo04, 2021-08-13:
Issues:

Profile switch is  buged.:
You can't change your personal profile say "Workday" to "Free Day.
If you do that, it was shown in NightScout as done, but AAPS is doing nothing and has still the old profile active.
To do the wanted profile switch, you have to change to the "Local Profile" first, then you can switch to the wanted one.
Because this Bug, the Automation  will also not work with profile switches.

Statistic:
The last fixes will not work for TDD.
All will shown now immediately,  but not TDD. 
TDD needs as long as before.

Disable Loop:
If you disable Loop, then you can't activate it again.
You have only the option to disconnect the pump. If you do that, you can activate the Loop again and reconnect the pump.

@MilosKozak 
If you have read that, you can close this issue. Wanted only that you are aware of this issues.

Philoul, 2021-08-16:
Statistic: with latest updates, on my phone Samsung A40), TDD is shown much quicker than previously (3 or 3 seconds later)

Disable Loop: I'll try to work on it today to propose a PR

Alamo04, 2021-08-16:
The most important is profile switch.
Because many users are using NightScout for overview at home and will then not recognise that the switch was not done. 

Also many are using Automation now. If they do a switch over it they didn't see this issue, until they use the phone and checked the overview in AAPS.

MilosKozak, 2021-08-17:
I tested profile switch functionality and it works well
It happened to me that profile was not saved correctly in NS (still old values after page refresh) thus not sent to AAPS. 2nd times it did work.
AAPS is a bit ahead to NS. NS still cannot distinguish between PS and effective PS thus NS may provide false data based on expectation that PS was already executed

Alamo04, 2021-08-17:
@MilosKozak 
Then I can't understand that, because I tested it just now.
My active Profile is "Freier Tag" and I copy that into "Freier Tag copy", than I have done a profile switch to this new profile.
As I wrote above, the profile switch was shown in NS and also in Treatments, but AAPS is doing nothing.

You can see that also in the pump tab, nothing happened here directly after profile switch and AAPS must change the profile in the pump too. Also the main page shows the old profile name.

I can handle it, because I aware of that and all profile switches works, if you first activate the standard profile, named "Local Profile"

[image]
[image]

Alamo04, 2021-08-17:
Thanks @Philoul ,

Can you do me a favour please and test the profile switch, as described above ?

Only want to know if it is a problem on my side, or a real Dev problem.

Philoul, 2021-08-17:
On my side all my PS worked, but I only use NS to share my BG values with my dev phone, and I don't have PS in my current automation rules.
I can make some tests in my dev phone but I didn't understand exactly what's wrong or which test I have to do 🤔...

Alamo04, 2021-08-18:
@Philoul 
Test:
- First at all use local profile in AAPS instead of NS.

- Make two local profiles (you can simply copy one) and name them ...say "Workday" and "Weekend".

-Then activate the "Workday" profile. This should work, because you are coming from standard profile.

- Wait a few minutes, so AAPS have time to send the new settings to the pump.

- Then try to change this profile to the other one (Weekend).

- Check if AAPS did this switch. You can see that on the main page and in Treatments.  In my case they are done in NS only, as you see in the pictures above.

If AAPS did this switches between this two profiles, this issue comes from my system.
If not, it comes from AAPS and must be fixed.

Philoul, 2021-08-18:
@Alamo04 Ok I'll test today one more time on my dev phone...
See a screenshot of an extract of previous PS done, I allways 2 lines with NS and PS (PS made between different profiles, with or without duration, percentage or timeoffset...)
[image]

Can you add the correspondant screenshot of User Entries ? (latest green icon on the right of treatments) may be it could help for understanding something...

Alamo04, 2021-08-18:
[image]

Only the switch between 2 different profiles dosn't work, if not one of them is the standard profile.
A switch to set the profile % higher, or deeper and then switch back to 100%, is working fine.

Alamo04, 2021-08-18:
@Philoul 
I tryed a complete clean and new Installation of AAPS.
- Deinstallation of AAPS
- Phone restart
- New Installation of AAPS
- Phone restart

Same issue with Profile, as you see in attached Screenshots.

Every try to activate "Arbeitstag" or "Freier Tag" failed and was only sended to NS as done.

Only if I use the "Lokales Profile" and then "Freier Tag" both are done in 
NS and also in AAPS.

[image]

[image]

I hate to write with my Smartphone.😅

Philoul, 2021-08-18:
Hello
I made other tests on my dev phone (low res) and it works (my loop and dev phones are both connected with an Insight pump)
=> Profile Switch standard, with percentage, with duration, timeshift... all worked fine for me

@MilosKozak could it be a problem linked to pump driver or connexion issue during profile upload ? (`setNewBasalProfile(Profile profile)` not succeed?)

[image]

Alamo04, 2021-08-19:
@Philoul 
Thanks for testing.

Then it has to do with my system, or the Medtronic driver, but the system works fine with the older versions, before database and driver are changed.

Will call a friend to check it with his Medtronic and I tell Andy about that.

Also, I have an Spirit Combo here, so I will test it with that pump first.

Alamo04, 2021-08-19:
@Philoul 
OK, it isn't the pump driver because I tested it now also with the Accu Check Combo, with same results.
AAPS didn't try to change the profile in the pump and didn't change it in AAPS.
AAPS changed the profile only in NS.

**Yiehhhaaa.... found the reason.** 

- This issue happens only if all settings in the profiles are the same and only the target ranges (times) are different. 

- If you make some small changes to the Basal rate too, in one profile, the switches are working fine.

Hope you can confirm that. 
I changed my "Free Day" profile basal rate only from 16,28 daily Basal to 16,30 daily basal.
My "Workday" profile stays at 16,28 and now all the switches works. 

By the way:
Another smal issue in Local Profiles are still remaining. 

All values of the daily Basal, in all profiles, are showing the same number.
This is the daily basal number of the last saved, or resettet profile.
It's only a visual issue.

Philoul, 2021-08-19:
Ok I saw your latest small issue concerning the daily basal of profile...
It should be easy to fix...

MichaelJohnSmith, 2021-08-19:
The problem changing profiles with only target deifferences seems related to
 #611 Target value not shown correctly on Home screen 
I was updating just the target of an existing profile then reloading it

Alamo04, 2021-08-19:
@MichaelJohnSmith 
Sure, because the profile was not set/changed from/in AAPS.
You didn't see that because it was the same Profile name.

Alamo04, 2021-08-22:
@Philoul 
Found another issue, but dont know if it was intend that AAPS works that way, or if it was a bug.

Autosense changed also your actual Target and was showing it on the main page, but only if you are using SMB Algorithm. 
It doesn't show that with AMA Algorithm, here you see only the Target that you have set in the profile, but it was changed by Autosense.

Should that work that way ?

Philoul, 2021-08-22:
@Alamo04 I'm not an expert of this part of code, but If I remember well, when this functionality was added, it was associated with openAPS SMB algo... So designed that way.

Alamo04, 2021-08-22:
Good to know.
That makes me sad, because I and many other did use AMA instead of SMB.

So AMA user have to life with the wrong Target shown. 😓

Philoul, 2021-08-23:
@Alamo04 I think it's not a wrong target shown, It's just the target set on profile and used by openAPS AMA for regulation.
With openAPS SMB, It's not the same algorithm and there is an auto adjustment of target to be more or less agressive according to sensibility, It's not the same algorithm... So you have to select openAPS SMB for this feature.

Alamo04, 2021-08-23:
@Philoul 
The target was set from autosense in both, SMB and AMA. I don't see any differences here.

My target is just now set to 90 and this is what I see on main page. But in real it is 115, because autosense has changed it.
Autosense is set to 1,5/0,5

I don't like SMB, because I lost my 14 day overview over my daily basal, and also I dont want to set my basal to zero, with every meal.

Philoul, 2021-08-23:
Ok, I saw the setting in AMA, it's posible that there is a bug... (as I told you, I'm not a specialist of this part)
During meal, basal is replaced by SMB... (more efficient).

Alamo04, 2021-08-23:
I know the reason, but with every SMB (Bolus), you lost the overview over your real given basal.
Thats why I don't use SMB, because my basal changed during the year.

I was also unsure if it is a bug, or not.

Alamo04, 2021-09-15:
@MilosKozak 
Dont want to make a new Issue , so I write it here.

**Latest Dev destroys Automation:**
- All Action settings are deleted 
- If you try to make a new Action, it is not possible. Perhaps this Action was created with a standard value and if you tab on it, to change it, an empty window was opened.

Philoul, 2021-09-15:
@Alamo04 I made a fix for the automation issue that is merged in latest version of dev (it will probably be necessary to import settings with a backup made before the bug...)

Alamo04, 2021-09-15:
@Philoul 
Works now, sorry didn't saw that you fixed it today. 👍😅

Alamo04, 2021-09-16:
@Philoul 
**Only for information,  this issue still exists:**
All values of the daily Basal, in all local profiles, are showing the same number.
This is the daily basal number of the last saved, or resettet profile.
It's only a visual issue.

But Profile switches are working now. 🙂

Philoul, 2021-09-16:
> Only for information, this issue still exists:
> All values of the daily Basal, in all local profiles, are showing the same number.
> This is the daily basal number of the last saved, or resettet profile.
> It's only a visual issue.

@Alamo04 Yes I know. I started to work on it but didn't push anything because it's not finished (currently you have the sum of default profile whatever profile selected...)
I have a version that updates the sum each time you select or save a profile but it doesn't work during edition. I think it's better to have the value updated in real time when you edit your profile...

Philoul, 2021-09-18:
@ Alamo04 PR made for Total Basal if you want to test it (This was my last job on this issue, I'd rather let Milos fix what's left)

Alamo04, 2021-09-18:
@Philoul 
Thanks, will test after merge.

Unfortunately it seems that the profile switch still didn't work, if the basal rates are not changed.
It looks like it works now, but I forgot that I have some differents in the Basal rate.

New Test with the same profile and changing only the ISF factors results in the old problem. Activation didn't work.

Perhaps it works after your fix too.

Philoul, 2021-09-19:
@Alamo04 
> Unfortunately it seems that the profile switch still didn't work, if the basal rates are not changed.
> It looks like it works now, but I forgot that I have some differents in the Basal rate.

I made following test with latest dev (2.8.2.3-dev) and it works:
- I made a copy of an existing profile (from Local Profile Plugin)
- I just changed one IC value and one ISF value (so all all basal rates are identical)
- Profile switches between the 1rst profile and the copy works fine

(test made with an insight pump)

Alamo04, 2021-09-19:
@Philoul 
Confirmed, works now. 👍

Thanks that you fixed it. 🙂

Alamo04, 2021-09-19:
@Philoul 
Confirmation refused. 

I'm using Milos last merge to 2.8.2.3

At my first test it works, why ever. 

Now, a few minutes ago, I wanted to change a few ISF values, saved the profile and activate it.
- same as before, changes are not done in AAPS, only in NightScout. 

Then I changed some ISF and IC, saved the profile and activate it.
- same as before, changes are not done in AAPS, only in NightScout. 

Then I copy that profile and make profile switches.
- same as before, changes are not done in AAPS, only in NightScout. 

Only way to get a profile switch in AAPS, or to reactivate your actual profile, with changes in ISF/IC, is still a different basal rate.

First after changing one basal rate, the profile switches between this two profiles works. The same if you want to male changes in ISF/IC in your actual profile and activate that.

I can only fix that, if I made a profile switch to a profile with a different basal rate and immediately switch back to the profile I want.

MilosKozak, 2021-09-20:
for me it works everytime i try it

MilosKozak, 2021-09-20:
TDD calculation takes time. it's lot of data. But it should not be blocking UI

Alamo04, 2021-09-20:
> for me it works everytime i try it

OK, I have tryed a lot of thinks and it's not working for me.
If I not change the Basal, then the profile was only  in NS, activated, not in AAPS.

I can change ISF and/or IC factors in my actual profile, then I can save it. If I now activate it, with the new settings, this activation is only done in NS.
If I make a smal change to the Basal too, the activation was also done in AAPS.
[image]

Alamo04, 2021-09-20:
@MilosKozak 
With your last merge, the App is broken.
After installing it and start AAPS, it crashed while pump status (Medtronic 754) is reading on page 2/4.

The merge before is still working.
It includes only:
- https://github.com/nightscout/AndroidAPS/commit/77e3c736085a99c36c73eee0c3d915caa13eba47
- https://github.com/nightscout/AndroidAPS/commit/cb2402ae453f5a4b6c41af7111863a3f5d603bbb
- https://github.com/nightscout/AndroidAPS/commit/02b77ae554236b74c359b92eb7d9018a2730240a
-

MilosKozak, 2021-09-21:
fixed by https://github.com/nightscout/AndroidAPS/commit/9e42f91848ba985bf67d084c854ae19245a89332

MilosKozak, 2021-09-21:
closing this issue as it seems to be resolved
for profile switch:
* make sure you have all "PUMP" logging enabled
* make this happen again
* create new issue with logs at least 5 min before and 15 min after

PR #634 (2021-09-15): Fix Overview LoopDialog (Disable Loop)
I tried to simplify User Interface of Loop Button:
When Loop is Disabled => Only Enable Loop Button
When Loop is Suspended => Only Resume Loop Button
When Pump is Disconnected => Only Reconnect Pump Button

I fixed #630 concerning "Disable Loop" in Loop Dialog, but code still remains complicated (I hope I didn't generate side effects...)
I'll try to see if I can simplify more...
closed 2021-09-21 by MilosKozak