nightscout/AndroidAPS #631: Inaccuracy in COB leading to a possible insulin overdose.

bchris21, 2021-08-14:
Hello,

I am using NSClient to monitor my son while he is in kindergarten. I have realized the following issue which I would like to report:

**Issue:**
COB between AAPS and NSClient are inaccurate and not synced.

When does it happen:
Whenever I manually delete carbs in AAPS from Treatments tab. The carbs deletion does not sync in NSclient and COB is different including the extra carbs added.

**Why it can be dangerous:**
When my son is in kindergarten and I use Bolus Calculator (in NSclient) before sending SMS bolus, NSClient miscalculates bolus as it counts more carbs on board thus suggests more insulin and risking an overdose.

**Workaround:**
I have to manually check deleted carb entries in AAPS and manually remove them from NSClient Treatments' tab.

**App Versions/Builds:**
AAPS 2.8.2 Build: c07465b0c-2021-03.07-00:56 Fullrelease
NSClient 2.8.2 Build: 194549adc-2021-01-23-13:06 nsclientfullrelease

**Attachments:**
[image]
[image]
[image]
[image]

I replicated the issue at 14/08/21 13:20 CET 

Logs from both AAPS and NSClient are attached.

[AndroidAPS_LOG_1628940043853.log.zip](https://github.com/nightscout/AndroidAPS/files/6986371/AndroidAPS_LOG_1628940043853.log.zip)
[AndroidAPS_LOG_1628940082178.log.zip](https://github.com/nightscout/AndroidAPS/files/6986372/AndroidAPS_LOG_1628940082178.log.zip)

Any help would be appreciated.

Thanks, 

Chris

MilosKozak, 2021-08-17:
can you send screenshots with COB graphs when you see this issue again?

bchris21, 2021-08-17:
@MilosKozak below the Screenshots requested.

**In detail:**
- I added fake 10gr at 23:57 to replicate the issue.
[image]

- I waited 5mins for the COB graph to update and took screenshots on both apps.
_AAPS Screenshot_
[image]

_NSClient Screenshot_
[image]

- Then I removed the carbs from AAPS's treatment tab but still carbs are shown on NSClient and started getting "consumed".

_AAPS Screenshot_
[image]

_NSClient Screenshot_
[image]

MilosKozak, 2021-08-19:
so the problem is only when you you delete carbs?

bchris21, 2021-08-19:
Exactly. Carbs addition syncs ok, but once carbs are deleted they do not sync on NSClient and they are accumulating until consumed by the pass of time or by BG drop.

MilosKozak, 2021-09-20:
is the entry removed from NS?

bchris21, 2021-09-24:
Hi Milos, sorry for the late reply. Just replicated the issue and I can confirm that carbs are removed from NS on the next BG update.