nightscout/AndroidAPS #637: 2.8.2.2 Dev : can't restart loop if stopped

robertrub, 2021-08-20:
In 2.8.2 main, I can stop and restart the loop from the main page. In 2.8.2.2 dev, if I stop the loop, I need to go in the settings to reactivate the loop option.
When it is activated, I have a very strange 364days (deactivated ?) under the button.
[image]
[image]
[image]
[image]

robertrub, 2021-08-20:
Follow-up. It seems that from the user interface, I can only "downgrade" loop functions but not "upgrade". If I select _open loop_, then I can NOT select _closed loop_ from the main page. I need to go to the settings to activate closed loop. 
[image]
[image]

MichaelJohnSmith, 2021-08-20:
You can also select Low Glucose Suspend.  Then click the loop icon again to select Closed Loop.
Not correct, but slightly easier than going to settings!

robertrub, 2021-08-20:
Nice try but no, it doesn't work of you turn off the loop. ;)

I never turn off the loop but I'm in testing stuff mode so I hit all the buttons available  :)

Philoul, 2021-08-20:
I made a PR to fix bug of disable loop...
Concerning the direct switch from open loop to closed loop,  I have to look into constraints (closed loop is currently not aloud from open loop, even if relative objectives are passed...)

Philoul, 2021-08-20:
I also prepared a commit (not currently pushed in the PR) to improve a bit user interface when loop is suspended or pump disconnected:
It will be possible to resume (respectively reconnect pump), or select a new duration.
My computer is off for 3 days, I'll update PR when I'll back home, but you can test current proposal and add comment if necessary...

MilosKozak, 2021-09-15:
should be resolved now

closed 2021-09-15 by MilosKozak

CaptainBalou, 2021-09-15:
That works for me. I can disconnect and reconnect my pump. In addition I can see that I'm now able to extend the disconnection time while I'm already disconnected. This is awesome!

robertrub, 2021-09-16:
Hi Milos. It is partially solved. I can select "open loop" and come back to "closed loop" but I cannot stop the loop and then restart it from the main UI. Once the loop is stopped, I need to go in the config generator in order to reactivate the loop. I don't know if this is a bug or a feature ;)
Robert

Philoul, 2021-09-16:
@robertrub can you share screenshot and explain the exact sequence of actions done from loop enabled to situation with issue ?
If I can reproduce it I'll be able to solve it...
Thanks

robertrub, 2021-09-16:
[image]
Above is from 2.8.2 Main. It is working ok.

https://user-images.githubusercontent.com/76626474/133663018-a7f6b787-076e-4c28-ba11-0dd7bc61031a.mp4

Above is from the latest dev. If I stop, it will only restart fron config.
[image]

Philoul, 2021-09-16:
@robertrub if you test Dash branch you don't have latest commit included in dev branch yesterday (wait a little bit you'll probably have this fix soon...)

robertrub, 2021-09-16:
Ok. If @Philoul says that it is fixed, them it is ;) I'd thought that DASH dev had the latest of dev...