nightscout/AndroidAPS #639: 2.8.2.2 Dev : Insulin delivered message

robertrub, 2021-08-20:
DASH driver (but same bug is also present in 2.8.2 main and EROS driver). 
The upper part says x U **will be** delivered. That message is ok. 
The lower part says x U _delivered_ when the x U has not been delivered yet.
Either show the real amount delivered (best solution) or a percentage of the sent amount (not very usefil) or delete this lower message (easiest solution).
[image]

Philoul, 2021-08-20:
The message included in Insight driver is the best I think (on each update of horizontal bar graph we have an update of text
"0.8 U/ 5.95 U has been delivered"
What do you think of moving some of the common pump strings in core to standardize User Interface and translation, and only keep in pump module specific strings?

MilosKozak, 2021-09-15:
@Philoul does omni have this information?

robertrub, 2021-09-16:
No, the same "error" is in Omni too. It says the same _xU (the full amount) have been injected_ when it is still in the process of injecting.

MilosKozak, 2021-09-20:
resolved in https://github.com/nightscout/AndroidAPS/commit/77e3c736085a99c36c73eee0c3d915caa13eba47

closed 2021-09-20 by MilosKozak