nightscout/AndroidAPS #640: [2.8.2.2 - dev] ISF lower than 2 mmol/l results in invalid profile

smokkelaar, 2021-08-22:
When setting the ISF lower than 2. it will not validate the profile.

[image]
[image]

It was still working correctly in 2.8.2 main.

Couldn't find the exact build where it goes wrong.
Multiple people on discord could validate this issue.

DM-LTR, 2021-08-24:
Same issue here with 2.8.2.2 dev - when ISF is set below 2.0, profile cannot be saved.

MilosKozak, 2021-09-15:
https://github.com/nightscout/AndroidAPS/commit/a448efe09cfeb7c54e28c5d4c67783d30fc6d918

closed 2021-09-15 by MilosKozak