nightscout/AndroidAPS #645: [2.8.2.2 Dev] Sometimes manual bolus are not correctly recorded in database

Philoul, 2021-08-24:
After several month of tests, I found about one week ago a very strange bug with manual bolus (done with "Insulin Button" on main screen). I saw this bug 3 times, 2 times last week, and last time this morning at 9h59.
(Hopefully the great majority of boluses are correctly recorded)

This morning art 9:59, I made a manual bolus with Insulin button, this bolus was correctly delivered (I saw the Pop Up with the bar graph), but later when I wanted to make another bolus, I saw that previous bolus made at 9h59 was missing in IOB graph and in Treatment tab, I could only see it in User Entry and in the history of my Pump ???
- I tried to analyse Log, but and don't see anything that could explain why this bolus is missing in treatments and IOB calculation
- In NS/Synchronization, I have only "Upload data to NS" enabled, all other settings are disabled
- - Pump Accu-Check Insight

[image]

I attached an Excel file with concatenation of log files and extract of each on of the 3 boluses made this morning:
- First tab with complete log (I only removed D/GLUCOSE and D/AUTOMATION to limit the number of lines)
- Second Tab Bolus 10.51U made at 08:04 with QuickWizard, (everything is OK)
- Third Tab Bolus 4U made at 09:59 with Insulin Button, correctly delivered in pump but not shown in Treatments, IOB calculation and IOB graph
- Fourth Tab Bolus 4U made at 11:15 with Insulin Button (everything is OK)

[LogBolusKO.xlsx](https://github.com/nightscout/AndroidAPS/files/7041758/LogBolusKO.xlsx)

overfrenk, 2021-08-24:
It happened to me 2 times too! I had lunch and I used the insulin and carbohydrate button, the bolus was correctly delivered and the carbohydrates were correctly registered but the insulin was not! In fact aaps had started doing 1U boluses via SMB and I noticed it. To solve the problem I then did a no delivery bolus to block SMB.

Philoul, 2021-08-24:
@overfrenk which pump?

overfrenk, 2021-08-25:
@Philoul accu-check insight

Philoul, 2021-08-25:
Thank's
I don't know if it's in the driver or not but I also have an Insight pump...
I propose to add logs and database verification in driver to help me...

MilosKozak, 2021-09-15:
@Philoul all database records (pump sync) are logged in dev version
@overfrenk @Philoul can you check logs when this happen again?
i didn't see this issue with Dana pumps

Philoul, 2021-09-16:
@MilosKozak the log is in the first tab of attached Excel file (it includes the bolus not recorded surounded by 2 boluses Ok, and a pre analysis on different tabs). It's easier for me to filter in excel, but I didn't find what could be wrong.
I'll try to look deeper in log with D/DATABASE records and pumpsync.

Philoul, 2021-09-18:
@MilosKozak See below the 3 log of Database record (extract from attached Excel File) corresponding to the 3 Bolus made
=> In the middle the missing Bolus in my screenshot (I don't see any differences in these lines...)

08:05:27.828 [Thread-8135] D/DATABASE: [PumpSyncImplementation.syncBolusWithPumpId():175]: Updated Bolus Bolus(id=1362, version=2, dateCreated=1629785127809, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=61248bfc60a0750004199cd4, pumpType=ACCU_CHEK_INSIGHT_BLUETOOTH, pumpSerial=33013206, temporaryId=null, pumpId=7226, startId=null, endId=null), timestamp=1629785074781, utcOffset=7200000, amount=10.51, type=NORMAL, isBasalInsulin=false, insulinConfiguration=null)

10:00:07.340 [Thread-10905] D/DATABASE: [PumpSyncImplementation.syncBolusWithPumpId():175]: Updated Bolus Bolus(id=1410, version=2, dateCreated=1629792007310, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=6124a70360a0750004199d1b, pumpType=ACCU_CHEK_INSIGHT_BLUETOOTH, pumpSerial=33013206, temporaryId=null, pumpId=7242, startId=null, endId=null), timestamp=1629705586292, utcOffset=7200000, amount=4.0, type=NORMAL, isBasalInsulin=false, insulinConfiguration=null)

11:16:05.058 [Thread-91] D/DATABASE: [PumpSyncImplementation.syncBolusWithPumpId():175]: Updated Bolus Bolus(id=1431, version=2, dateCreated=1629796565023, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=6124b8cf60a0750004199d4a, pumpType=ACCU_CHEK_INSIGHT_BLUETOOTH, pumpSerial=33013206, temporaryId=null, pumpId=7251, startId=null, endId=null), timestamp=1629796543002, utcOffset=7200000, amount=4.0, type=NORMAL, isBasalInsulin=false, insulinConfiguration=null)

Excel file containing the log containing the 3 boluses (filtered with pumpsync event, bolus OK in green, bolus KO in orange)
[LogBolusKO.xlsx](https://github.com/nightscout/AndroidAPS/files/7190610/LogBolusKO.xlsx)

Philoul, 2021-09-18:
@MilosKozak I had this issue again today just after midnight

See attached log (between 23:52 and 00:11)
=> The Bolus and the two SMB after are recorded in User Entries and in my pump history, but not recorded in meal/bolus tab and not shown in overview graphs
=> In log the 2 boluses (8U and 0.70U) seems to be correctly recorded in database (the last one (0.25U) is after the end of the log file
=> Previously I only noticed this bug with Insulin button, but we see that we can also miss some SMB too

[image]

[AndroidAPS._2021-09-18_14-05-04_.zip](https://github.com/nightscout/AndroidAPS/files/7190742/AndroidAPS._2021-09-18_14-05-04_.zip)

MilosKozak, 2021-09-20:
```
00:01:28.991 [Thread-9389] D/DATABASE: [PumpSyncImplementation.syncBolusWithPumpId():174]: Inserted Bolus Bolus(id=6009, version=0, dateCreated=1632002488948, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=ACCU_CHEK_INSIGHT_BLUETOOTH, pumpSerial=33013206, temporaryId=null, pumpId=9072, startId=null, endId=null), timestamp=1632002488746, utcOffset=7200000, amount=8.0, type=NORMAL, isBasalInsulin=false, insulinConfiguration=null)

....
....

00:01:44.871 [main] D/UI: [BolusProgressDialog.onResume$lambda-4():132]: Status: 3,20 U / 8,00 U délivrés Percent: 40
00:01:44.978 [RxCachedThreadScheduler-62] D/CORE: [DataBroadcastPlugin.sendBroadcast():203]: Sending broadcast info.nightscout.androidaps.status to: sk.trupici.g_watch
00:01:46.853 [main] D/UI: [ActivityMonitor.onActivityPaused():36]: onActivityPaused: BolusProgressHelperActivity elapsed=26216 total=13889116
00:01:46.860 [main] D/UI: [BolusProgressDialog.onPause():159]: onPause
00:01:46.946 [RxCachedThreadScheduler-67] D/DATABASE: [CompatDBHelper.dbChangeDisposable$lambda-18():70]: Firing EventTherapyEventChange
00:01:46.959 [Thread-9389] D/DATABASE: [PumpSyncImplementation.insertTherapyEventIfNewWithTimestamp():222]: Inserted TherapyEvent TherapyEvent(id=1118, version=0, dateCreated=1632002506861, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=ACCU_CHEK_INSIGHT_BLUETOOTH, pumpSerial=33013206, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1632002505849, utcOffset=7200000, duration=0, type=NOTE, note=Pompe suspendue, enteredBy=AndroidAPS, glucose=null, glucoseType=null, glucoseUnit=MGDL)
00:01:46.984 [RxCachedThreadScheduler-48] D/DATABASE: [CompatDBHelper.dbChangeDisposable$lambda-18():70]: Firing EventTherapyEventChange
00:01:46.985 [Thread-9389] D/DATABASE: [PumpSyncImplementation.insertTherapyEventIfNewWithTimestamp():222]: Inserted TherapyEvent TherapyEvent(id=1119, version=0, dateCreated=1632002506978, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=ACCU_CHEK_INSIGHT_BLUETOOTH, pumpSerial=33013206, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1632002505976, utcOffset=7200000, duration=0, type=NOTE, note=Maintenance M21 : Réservoir vide, enteredBy=AndroidAPS, glucose=null, glucoseType=null, glucoseUnit=MGDL)
00:01:47.021 [main] D/UI: [ActivityMonitor.onActivityResumed():41]: onActivityResumed: InsightAlertActivity
00:01:47.115 [Thread-9389] D/DATABASE: [PumpSyncImplementation.createOrUpdateTotalDailyDose():411]: Inserted TotalDailyDose TotalDailyDose(id=31, version=0, dateCreated=1632002507113, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=ACCU_CHEK_INSIGHT_BLUETOOTH, pumpSerial=33013206, temporaryId=null, pumpId=324255, startId=null, endId=null), timestamp=1631919600000, utcOffset=7200000, basalAmount=6.35, bolusAmount=65.48, totalAmount=0.0, carbs=0.0)
00:01:47.722 [Thread-9389] D/PUMPQUEUE: [CommandBolus.execute():28]: Result success: true enacted: true
````

There was an pump error during delivery and real delivered amount was not stored to DB at this time
then log ends. It's necessary to look more to the future in logs and look into DB for current state of Bolus(id=6009)

MilosKozak, 2021-09-20:
And Insight driver should not report `Result success: true` in this case

Philoul, 2021-09-20:
Concerning the bolus of 8U, it started before the cartridge replacement, (less than 8U remaing in reservoir) and finished after pump resume.
The 8U are correctly administered in my pump history.. 
I'll continue analysis hope I still have later log

Philoul, 2021-09-24:
I can now catch a bit more information with my additionnal logs and Notification (cf #688) 

[image]
```
pumpSync.syncBolusWithPumpId(
        startTimestamp,
        event.getImmediateAmount(),
        null,
        bolusID.getId(),
        PumpType.ACCU_CHEK_INSIGHT,
        serial);
aapsLogger.debug(LTag.PUMP, "XXXX Last bolus delivered " + bolusID.getId() + " Update database sent");
if (!checkLastBolusRecorded(startTimestamp, event.getImmediateAmount())) {
    aapsLogger.error(LTag.PUMP, "XXXX Last bolus delevered " + bolusID.getId() + " not recorded in database");
    Notification notification = new Notification(Notification.INFO, "Last Bolus delevered " + event.getImmediateAmount() + " not recorded in database", Notification.URGENT);
    rxBus.send(new EventNewNotification(notification));
}
```

```
private boolean checkLastBolusRecorded(long startTimestamp, double immediateAmount) {
    PumpSync.PumpState.Bolus lastRecordedBolus = pumpSync.expectedPumpState().getBolus();
    return lastRecordedBolus == null ? false : lastRecordedBolus.getTimestamp() == startTimestamp && lastRecordedBolus.getAmount() == immediateAmount;
}
```

And see extract of the SMB OK and SMB KO, everything seems to be the same except that `checkLastBolusRecorded` returns false

```
23:34:00.833 [Thread-9616] D/DATABASE: [PumpSyncImplementation.syncBolusWithPumpId():176]: Inserted Bolus Bolus(id=6727, version=0, dateCreated=1632519240769, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=ACCU_CHEK_INSIGHT_BLUETOOTH, pumpSerial=33013206, temporaryId=null, pumpId=9359, startId=null, endId=null), timestamp=1632519240577, utcOffset=7200000, amount=0.11, type=SMB, isBasalInsulin=false, insulinConfiguration=null)
23:34:02.598 [Thread-9616] D/PUMP: [LocalInsightPlugin.processBolusDeliveredEvent():1450]: XXXX Delivered Bolus 9359!=34776/326896 already recorded in Insight database
23:34:02.690 [Thread-9616] D/DATABASE: [PumpSyncImplementation.syncBolusWithPumpId():177]: Updated Bolus Bolus(id=6727, version=1, dateCreated=1632519242652, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=ACCU_CHEK_INSIGHT_BLUETOOTH, pumpSerial=33013206, temporaryId=null, pumpId=9359, startId=null, endId=null), timestamp=1632519241593, utcOffset=7200000, amount=0.11, type=SMB, isBasalInsulin=false, insulinConfiguration=null)
23:34:02.692 [Thread-9616] D/PUMP: [LocalInsightPlugin.processBolusDeliveredEvent():1463]: XXXX Last bolus delivered 9359 Update database sent
23:34:02.876 [Thread-9616] D/PUMP: [LocalInsightPlugin.processBolusProgrammedEvent():1380]: XXXX Bolus Programmed 9359/326895/326896 already recorded in Insight database
23:34:15.660 [RxCachedThreadScheduler-37] D/NSCLIENT: [NSClientPlugin.onStart$lambda-4():127]: DBADD treatments Sent PairBolus {"eventType":"Correction Bolus","insulin":0.11,"created_at":"2021-09-24T21:34:01.593Z","date":1632519241593,"type":"SMB","isValid":true,"isSMB":true,"pumpId":9359,"pumpType":"ACCU_CHEK_INSIGHT_BLUETOOTH","pumpSerial":"33013206"} 6726/6728
23:34:16.846 [RxCachedThreadScheduler-69] D/DATABASE: [NSClientAddAckWorker.doWork$lambda-9():127]: Updated ns id of Bolus Bolus(id=6727, version=1, dateCreated=1632519242652, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=614e4458d74db300042d557f, pumpType=ACCU_CHEK_INSIGHT_BLUETOOTH, pumpSerial=33013206, temporaryId=null, pumpId=9359, startId=null, endId=null), timestamp=1632519241593, utcOffset=7200000, amount=0.11, type=SMB, isBasalInsulin=false, insulinConfiguration=null)

00:04:08.796 [Thread-10061] D/DATABASE: [PumpSyncImplementation.syncBolusWithPumpId():176]: Inserted Bolus Bolus(id=6730, version=0, dateCreated=1632521048685, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=ACCU_CHEK_INSIGHT_BLUETOOTH, pumpSerial=33013206, temporaryId=null, pumpId=9360, startId=null, endId=null), timestamp=1632521048480, utcOffset=7200000, amount=0.11, type=SMB, isBasalInsulin=false, insulinConfiguration=null)
00:04:10.579 [Thread-10061] D/PUMP: [LocalInsightPlugin.processBolusDeliveredEvent():1450]: XXXX Delivered Bolus 9360!=34777/326903 already recorded in Insight database
00:04:10.698 [Thread-10061] D/DATABASE: [PumpSyncImplementation.syncBolusWithPumpId():177]: Updated Bolus Bolus(id=6730, version=1, dateCreated=1632521050648, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=ACCU_CHEK_INSIGHT_BLUETOOTH, pumpSerial=33013206, temporaryId=null, pumpId=9360, startId=null, endId=null), timestamp=1632434649577, utcOffset=7200000, amount=0.11, type=SMB, isBasalInsulin=false, insulinConfiguration=null)
00:04:10.703 [Thread-10061] D/PUMP: [LocalInsightPlugin.processBolusDeliveredEvent():1463]: XXXX Last bolus delivered 9360 Update database sent
00:04:10.936 [Thread-10061] E/PUMP: [LocalInsightPlugin.processBolusDeliveredEvent():1465]: XXXX Last bolus delevered 9360 not recorded in database
00:04:10.981 [Thread-10061] D/PUMP: [LocalInsightPlugin.processBolusProgrammedEvent():1380]: XXXX Bolus Programmed 9360/326902/326903 already recorded in Insight database
[... cut ...] dId=null), timestamp=1632434649577, utcOffset=7200000, amount=0.11, type=SMB, isBasalInsulin=false, insulinConfiguration=null)
```

Log with 3 SMB not recorded
[AndroidAPS._2021-09-25_00-00-00_.0.zip](https://github.com/nightscout/AndroidAPS/files/7228686/AndroidAPS._2021-09-25_00-00-00_.0.zip)

Log with SMB correctly recorded (23h34 0.11U) to be able to compare
[AndroidAPS._2021-09-25_00-00-00_.0.zip](https://github.com/nightscout/AndroidAPS/files/7228692/AndroidAPS._2021-09-25_00-00-00_.0.zip)

Philoul, 2021-09-24:
I got 5 missing SMB between midnight and 1:00, then after a stop and restart of AAPS, next SMB was correctly recorded.
[image]

MilosKozak, 2021-09-30:
@Philoul so where is the problem?

Philoul, 2021-09-30:
It seems that sometimes Database is not updated correctly after pumpSync.syncBolusWithPumpId.
(I added an alert each time the lastrecorded bolus is not the same than the one updated by syncBolusWithPumpId immediatly, this alert should never occurs, but I got it several times, and each time a SMB delivered by the pump was not recorded in database)

I recently added new debugs to get more informations and continue analysis, but everything seems to be ok in Insight plugin (I suspect an issue difficult to catch because rare, in pumpsync implementation or in database).

Until now I only got this alert on my loop phone (never on my dev phone running in parallel), and I can have several days without any missing records, and 2 or 3 following days with one to 3 missing SMB (no more)...

Philoul, 2021-10-01:
I think I got it with my latest debug:
```
00:02:38.121 [Thread-28694] D/PUMP: [LocalInsightPlugin.deliverTreatment():610]: XXXX deliverTreatment 9685 Update database sent: 01/10/2021 00:02 amount: 0.11 Result: true
00:02:39.805 [Thread-28694] D/PUMP: [LocalInsightPlugin.processBolusDeliveredEvent():1451]: XXXX Delivered Bolus 9685!=35109/329906 already recorded in Insight database
00:02:39.867 [Thread-28694] E/PUMP: [LocalInsightPlugin.processBolusDeliveredEvent():1458]: XXXX Last bolus in Database before update 01/10/2021 00:02 amount: 0.11
00:02:39.887 [Thread-28694] D/PUMP: [LocalInsightPlugin.processBolusDeliveredEvent():1466]: XXXX Last bolus delivered 9685 Update database sent: 30/09/2021 00:02 amount: 0.11 Result: false
00:02:39.951 [Thread-28694] E/PUMP: [LocalInsightPlugin.processBolusDeliveredEvent():1468]: XXXX Last bolus in Database after update 30/09/2021 22:57 amount: 0.12
00:02:40.019 [Thread-28694] E/PUMP: [LocalInsightPlugin.processBolusDeliveredEvent():1470]: XXXX Last bolus delevered 9685 not recorded in database
00:02:40.031 [Thread-28694] D/PUMP: [LocalInsightPlugin.processBolusProgrammedEvent():1381]: XXXX Bolus Programmed 9685/329905/329906 already recorded in Insight database
```
SMB correctly recorded in deliverTreatment
ID is found in readHistory during the process of processBolusDeliveredEvent from pump, but the calculated date from pump event is shifted one day before compared to real timestamp (strange because this calculation was not modified during update of the plugin for new database)

=> I'll fix this bug very quickly.

closed 2021-10-01 by MilosKozak