nightscout/AndroidAPS #664: [Medtronic] Automation-triggered profile % changes cause lost pump connections

rustymonkey, 2021-09-06:
It's possible to trigger MDT pump connection failures that persist (until AAPS and RL are restarted), by enabling automation rules that change the profile %. 

Tested by enabling automation rules that:
- Temporarily (for 5 minutes) increase the active profile % to 120% when BG > 10mmol (if change in BG > -.3)
- Temporarily (for 5 minutes) increase the active profile % to 140% when BG > 15mmol (if change in BG > -.3)
The rules would trigger quite often (e.g. for a forgotten bolus or during a persistent high), and occasionally would result in the lost pump connection. During periods that the rules didn't trigger (or if the rules were disabled), connection stability was solid.

Upon changing the automation rules' duration to 10 minutes (instead of 5 minute durations), the bug still occurred (though less frequently).

Since the bug doesn't occur every time the rule is triggered (nor every time that the rule is triggered and a microbolus is given), I'd guess that it's some combination of the profile change + microbolus + temp basal that triggers the issue. I've submitted a log of the problem (direct to @andyrozman and to the default log address) that hopefully reveals the trigger. Note that the log was generated immediately after the bug occurred.

Notes:
- Tested with https://github.com/andyrozman/AndroidAPS/tree/hotfix_mdt_2_8_2 with an MDT 554 (mysentry disabled) / Nexus 5x|Android 8 / OrangeLink fw v2.5
- This issue seems to be distinct from #474 (RileyLink Initialization failures) which doesn't occur with the configuration above, but does occur if an S8 is used instead of the Nexus 5x even with automation disabled.
- This issue also seems to be distinct from https://github.com/MilosKozak/AndroidAPS/issues/2121 (failed reconnect after lost MDT connection) as I was unable to replicate issue 2121 with the above build by bringing the pump and/or RileyLink out of range of the phone until a disconnection occurred and then back in range. Also, when issue 2121 did occur for me (in earlier builds) automation rules triggering profile % changes weren't required to trigger the bug.
- I'm currently testing with an MDT 722 to see if the results are the same--I'll update the bug once I have enough data.

MilosKozak, 2021-09-08:
can you provide logs when this happen again?

rustymonkey, 2021-09-09:
Will do though it might take a few days (my son is currently out of town).

rustymonkey, 2021-09-17:
Update: I still don't have another good set of logs with the MDT 722 (since the original report, I've discovered that, at least with the 722, connection stability problems often arise when the OL battery starts getting low (the OL LEDs don't indicate that the battery has run out, but AAPS shows OL battery level of around 10%). I'm continuing to try to replicate the problem with the 722. If I can't, I'll try again with the 554.

That said, it might be a good idea for AAPS to improve reporting on OL battery status. i.e.:
- Display the status directly on the MDT tab next to the MDT battery status (rather than forcing the user to go another level deep to view the status in the RL stats page)
- Optionally display the status on the Overview page (i.e. next to pump battery, infusion set age, etc.)
- Warning when battery < x (15%?) (optional)

Alamo04, 2021-09-18:
@rustymonkey 
The Battery Status is shown on MDT Tab in dev versions, so I think it will come to the next Master too.

andyrozman, 2021-09-18:
Yeah this was added some time ago. First on Omnipod Eros driver and then few weeks later on Medtronic driver.

rustymonkey, 2021-09-20:
Looks good! 

fyi, I sent a Debug log that hopefully illustrates the issue (to the logs email & andyrozman)

andyrozman, 2021-09-21:
Where are you testing this? Master or dev? Can you reproduce it on dev?

rustymonkey, 2021-09-23:
I'm using: https://github.com/andyrozman/AndroidAPS/tree/hotfix_mdt_2_8_2

Alamo04, 2021-09-23:
@rustymonkey 
Why are you using such an old fix ?

Try the last Dev and see if something there isn't working.
https://github.com/nightscout/AndroidAPS/tree/dev

Alamo04, 2021-09-23:
@rustymonkey 
Don't understand why you are using such an old fix.

Use the AAPS Dev, there are all changes/fixes integrated. 
https://github.com/nightscout/AndroidAPS/tree/dev

Or use the Master, that's the one that is created for DIY Loop.
https://github.com/nightscout/AndroidAPS/tree/master

rustymonkey, 2021-09-23:
@Alamo04 
According to github for the build I'm using "This branch is 2 commits ahead, 1 commit behind nightscout:master. " Unless I'm misunderstanding, it seems to include an MDT fix that isn't in the current 2.8.2 release?

rustymonkey, 2021-09-30:
Quick update re. testing on the 722. As described above, AAPS 2.8.2 (with MDT hotfix) will generally work flawlessly for a week+, but once the bug occurs, it tends to recur every couple of days (I've submitted logs). 

What's I've since learned is that deleting the app cache solves the problem (so that AAPS is again working flawlessly).

MilosKozak, 2021-09-30:
@andyrozman do you have logs?

rustymonkey, 2021-11-10:
This issue has not occurred when testing AAPS 3.0 B3-B5 on an S7 running Android 11 (Pixel Experience) connecting to an MDT 554 via a RileyLink. 

Still to do: test with an OrangeLink (that'll take some time since I don't have a spare).

rustymonkey, 2021-11-14:
Tested AAPS 2.8.2.10 dev build 2021-11-11 11:40am on an S7 running Android 11 (Pixel Experience) connecting to an MDT 554 via an Orangelink and am no longer able to replicate this. Closing.

closed 2021-11-14 by rustymonkey