nightscout/AndroidAPS #667: Apps not responding while BG is changing. Had to press run now

margogoodfellow, 2021-09-08:
APPS did not respond for 20 min i had to press run now manually so it started working again. It worked fine for a few hours and it happened again. As you can see on the screenshot. Link to the logs attached. It happened at 14:30

https://drive.google.com/file/d/1wBriyCAbvbIbWKALabfkXQ_mdjoirgz2/view?usp=drivesdk

[image]
[image]

MilosKozak, 2021-09-13:
log doesn't contain timeframe with issue
TURN OFF AUTOSENS logging. it's huge and it's causing fast log rotation

margogoodfellow, 2021-09-13:
How do i turn it off please ? As I must be doing something wrong

MilosKozak, 2021-09-14:
Maintenance - Logging

margogoodfellow, 2021-09-14:
Thank you. I have done that now. I will have to wait when this happens
again to generate logs as i have deleted aps and reinstalled it and it
helped.

savek-cc, 2021-10-15:
@MilosKozak duplicate of https://github.com/nightscout/AndroidAPS/issues/598

margogoodfellow, 2021-10-15:
Nope the phone was charged and I have this option off. I use this phone for looping only . I had to delete the app and install it again and only this solved the problem.  I didn't have an issue since then but it wasn't first time when this happened   I have finally switch autosense and if it will occur again I will generate the logs

closed 2021-11-10 by MilosKozak

margogoodfellow, 2021-12-01:
https://drive.google.com/file/d/1r8zsKZe--5ifwzjb2SszRsNqc6ggyxH-/view?usp=drivesdk

Just had 15 min of loop not responding