nightscout/AndroidAPS #669: dev : aaps crash after install

MTR93600, 2021-09-12:
dev branch, last commit, the build works, but after install the apk, click on open button, aaps crash systematically.

Philoul, 2021-09-12:
I couldn't reproduce this crash on my loop phone, but I got it once on my dev phone the first time I built dev with OpenHuman refactoring (so I tried to analyse with the debugger...)

@MilosKozak and @TebbeUbben see below my analysis if it could help...
The problem seems to be in OpenHuman module in `OHCounterDelegate.kt` (line 12)
`private var value = sp.getLong("openhumans_counter", 1)`

Break point put on this line (I got an error when I tried to evaluate expression `sp.getLong("openhumans_counter", 1)`)

I just had to evaluate once `sp.putLong("openhumans_counter", 1)` and I could continue and AAPS starts on my dev phone without the crash
Unfortunatly I was not able to reproduce this crash after that...

I proposed @MTR93600 to test dev branch with the modification below and he could start AAPS without crash (but I'm afraid it's just a workaround, I don't think it's the root cause...)
```
private var value = try {
        sp.getLong("openhumans_counter", 1)
    } catch(e:Exception) {
        sp.putLong("openhumans_counter", 1)
        1L
    }
```

=> He still have a crash after testing this modification (???)
I hope this will help you...

MilosKozak, 2021-09-13:
related to this
https://console.firebase.google.com/u/0/project/androidaps-c34f8/crashlytics/app/android:info.nightscout.androidaps/issues/9a680d0fc839920f00aa2680d92ec3ff?versions=2.8.2.2-dev%20(1500)&time=last-seven-days&sessionEventKey=613CA45703DE00013D15CAE6F53F290F_1585101421281320186

MilosKozak, 2021-09-13:
I don't undestand how this could happen (crash is in android lib) but this should resolve it
https://github.com/nightscout/AndroidAPS/commit/dc90c6b140e9df77b5739ba1ccd32a951cc83bae

closed 2021-09-13 by MilosKozak

MTR93600, 2021-09-13:
i test the commit, but still crashing when I open aaps

MTR93600, 2021-09-14:
problem fixed :) thanks :)