nightscout/AndroidAPS #671: Last dev update pb activate profile

MTR93600, 2021-09-14:
when you activate the local profile, I get the message min target profile is out of hard limits
I try withy 90, 100, 110, 120
same message for all the value

MilosKozak, 2021-09-14:
paste screenshot of profile

MTR93600, 2021-09-14:
we are several. tester with the same bug
here the screenshots
[image]
[image]
[image]

PR #672 (2021-09-15): Fix ProfileSealed Target out of Hardlimit
Fix #671
closed 2021-09-15 by MilosKozak