nightscout/AndroidAPS #674: loop icon don't show after disconnect the pump

MTR93600, 2021-09-15:
if I disconnect the pump for 15 minutes, when I come back to close loop, the icon don't change, the time is still running like in overview the action didn't happen.
[image]

MTR93600, 2021-09-15:
At the end of the three minutes left, the icon change to loop

MTR93600, 2021-09-15:
it's in the last commit dev version

Philoul, 2021-09-15:
There are currently several bugs in "Loop Dialog" (I made a PR to fix them) :
"Reconnect pump" button missing when pump state is disconnected, "Enable Loop" button missing if loop is disabled...
=> In your Screenshot, I think you tried to reconnect your pump before the end of duration (and without Reconnect button in dialog it's difficult!)

@MilosKozak can you please take a look on #634 and tell me if it's ok for you or if you want some modification ? (This PR fix the bugs of current dev, improve the user interface compared to current master, and the code is simplified to be easier to maintain)

closed 2021-09-15 by MilosKozak