nightscout/AndroidAPS #685: [Feature Request] Automation: If HR and/or steps (from xdrip) then...

yossibivas11, 2021-09-23:
Xdrip has a feature of counting and displaying HR and steps "density". That would be a good indication for starting an "activity" TT as an automation.

reijoahola-nightscout, 2021-09-23:
xDrip+ can detect also walking and moving faster (driving ?). The feature is called "Motion detection" but it does not work in all phones.

yossibivas11, 2021-09-23:
I imagine an automation that can sense that I got on my bicycle and activate a TT. It could be high HR and fast movement or high step frequency.

swissalpine, 2021-09-25:
I don't expect much from such automation: if the activity TT is only set when the movement takes place, it is too late to have an effect on the IOB. For a functioning therapy adjustment, this usually has to be set an hour or more before.
And: not every heart rate increase, not every high step rate has to be countered with a TT. The quick sprint to the bus, climbing the stairs, etc. is usually too short to make a therapy adjustment necessary. However, if I observe heart rate or steps over a period of time first in order to set a TT, the adjustment comes much later (see the problem description above).

yossibivas11, 2021-09-25:
I disagree. If I forget to activate a high TT at the beginning of a ride, my BG drops from boli that I get during the ride, which could have been avoided with high TT.
Sometimes I forget to activate a TT when I get on the bike or start a walk (many hills around me) so I want a failsafe to salvage the situation- a combination of high step frequency and high HR for a whole ~minute could be a nice way to do it.

ffsb42, 2022-10-03:
about that topic:

- leveraging Heart Rate (HR) would be excellent to detect rest vs physical activity across the board; besides kids, adults do quite often get caught off guard in hypo due to unplanned physical activities (carrying heavier than anticipated groceries, walking stairs, running to catch a bus, sex,...)
- leveraging steps can be a bit harder as a number of trackers can be misleading (for instance watch trackers if you are hammering nails in a board....)
- integration on Apple is relatively simple through Apple Health. 
- integration on Android could probably be done through Google Fit or the new upcoming Google "_Health Connect by Android_"  https://play.google.com/store/apps/details?id=com.google.android.apps.healthdata&hl=en_US&gl=US
-

olorinmaia, 2026-09-23:
This is now possible in master. Heart rate and steps can be used as trigger for automations, collected from wear app. Closing issue.

closed 2026-09-23 by olorinmaia