nightscout/AndroidAPS #689: [Dev] [Bug]Intermittent Issue with Delta values not being passed from glucose_status to _determine_basal

tim2000s, 2021-09-24:
In the current Dev code, the values for Delta, Short Average Delta and Long Average Delta are not always passed to determine_basal. 

Screenshots attached showing example where none is showing up on either the GUI main screen or in openAPS SMB in spite of a change in glucose value of 32 mg/dl. Dev build listed in image. 

Logfile also attached. 

[image]
[image]
[image]
[image]
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/7226925/AndroidAPS.log)

dexdan, 2021-09-24:
Likely linked to https://github.com/0pen-dash/AndroidAPS/issues/104

MilosKozak, 2021-09-30:
it's caused by your wrong setup: doubled BG values with identical timestamp.
G6 is smoothing old values and you are probably synchnonizing them elsewhere which breaks it

closed 2021-09-30 by MilosKozak

tim2000s, 2021-10-02:
If there are dual glucose entries for each point, then the backfill from NS must be picking up additional data from Dexcom bridge in Nightscout which has the same value and a slight time delay. I'm surprised that the backfill from NS would pick this up and use it, but I think that explains the problem. 

I wonder how many others use dual feeds to NS for redundancy purposes?

MilosKozak, 2021-10-02:
I don't know your setup but for Dexcom patched app + aaps is recommended for NS upload

tim2000s, 2021-10-05:
> I don't know your setup but for Dexcom patched app + aaps is recommended for NS upload

Yes. Nightscout upload is configured via AAPS. There is also an upload via Share, which had the Bridge component enabled in Nightscout providing an alternative route. 

The key difference between master behaviour and Dev behaviour is that in the Dev version, the dual values are being recognised and generating incorrect delta data. In master, this wasn't happening. 

It looks like the Dev handles duplicate timestamps differently.

MilosKozak, 2021-10-05:
Double upload can cause only troubles. Keep only AAPS which handles backward smoothing properly

MilosKozak, 2021-10-05:
Previous versions didn't upload smoothed values to NS

vanelsberg, 2021-11-06:
@MilosKozak I do not agree on the way it seems to works now:

When in AAPS user **explicitly defines a BG source** (xDrip, or any other BG source then NS) it should ignore BG information from NS. This includes complaining about double values....

Imho AAPS version 2.8.2.9 is introducing a dependency to NS that is not needed or even unwanted:
User should be able to configure AAPS to not download *any* data from NS ("upload-only" as was available for 2.8.2)

Please see also https://github.com/nightscout/AndroidAPS/issues/857

vanelsberg, 2021-11-06:
@MilosKozak 
> I don't know your setup but for Dexcom patched app + aaps is recommended for NS upload
I do not see what this has to do with the Dexcom. Same goes for any other "BG source"

What you are saying is the AAPS does not accept *any* BG source writing values to NS even if it does not need or use these values? I can not see the point in this...