nightscout/AndroidAPS #692: Dev pb with Dana smb and tbr

MTR93600, 2021-09-26:
in some cases, the message appear in the loop page, error wrong value about the TBR. 
I tried to identify why, and the only thing I can observe was the pump run tbr at 0.19 when aaps want to send for the same % 0.17
When this appear, if with the TBR I have a SMB, the SMB will not be send to the pump, until the value become correct for the TBR. 
The pb fixe by himself later during the rise, until the glucose value become stable, and the error show up again. 
Never have this issue on the master.
I test with Dash too, and I didn't have the pb.

MTR93600, 2021-09-26:
to complet my explanation, every morning I disconnect the pump for the shower, and I got the every time the same message of alerte, error TBR. After that I have to force the connection to the pump several time. Sometime I can get the temp basal true in loop and dana page, sometime it's staying in error for wrong value in loop page.
[image]

Philoul, 2021-09-26:
I think it's probably an issue in pump driver.
I also disconnect my pump every morning for my shower (for about 15min), I sometimes reconnect it before the 15min, sometimes pump is automatically reconnected after 15min, but I never got this popup with my Insight.

MichaelJohnSmith, 2021-09-26:
I am using a Dana-i with the latest dev.  I also disconnect each morning and do not experience this problem.

MTR93600, 2021-09-26:
I'm using Dana-RS

MilosKozak, 2021-09-26:
provide logs when this happen again

MilosKozak, 2021-09-26:
be sure you have all PUMP* logging enabled

MTR93600, 2021-09-27:
12:21:01.547 [main] D/CONFIGBUILDER: [ConfigBuilderPlugin.storeSettings():64]: Storing settings from: CheckedCheckboxEnabled
12:21:01.548 [main] D/CONFIGBUILDER: [PluginStore.setFragmentVisibilities():126]: Selected interface: OpenAPS SMB
12:21:01.548 [main] D/CONFIGBUILDER: [PluginStore.setFragmentVisibilities():126]: Selected interface: Profil d'insuline ajustable Oref
12:21:01.548 [main] D/CONFIGBUILDER: [PluginStore.setFragmentVisibilities():126]: Selected interface: Sensibilité Oref1
12:21:01.549 [RxCachedThreadScheduler-54] D/CORE: [UserEntryLogger$log$2.invoke():42]: USER ENTRY: PLUGIN_ENABLED ConfigBuilder OpenAPS SMB [SimpleString(value=OpenAPS SMB)]
12:21:01.549 [main] D/CONFIGBUILDER: [PluginStore.setFragmentVisibilities():126]: Selected interface: Profil Local
12:21:01.549 [main] D/CONFIGBUILDER: [PluginStore.setFragmentVisibilities():126]: Selected interface: App Dexcom (patchée)
12:21:01.549 [main] D/CONFIGBUILDER: [PluginStore.setFragmentVisibilities():126]: Selected interface: Dana-i/RS
12:21:01.550 [main] D/CONFIGBUILDER: [ConfigBuilderPlugin.savePref():82]: Storing: ConfigBuilder_GENERAL_ActionsPlugin_Enabled:true
12:21:01.550 [main] D/CONFIGBUILDER: [ConfigBuilderPlugin.savePref():86]: Storing: ConfigBuilder_GENERAL_ActionsPlugin_Visible:true
12:21:01.550 [main] D/CONFIGBUILDER: [ConfigBuilderPlugin.savePref():82]: Storing: ConfigBuilder_INSULIN_InsulinOrefRapidActingPlugin_Enabled:false
12:21:01.551 [main] D/CONFIGBUILDER: [ConfigBuilderPlugin.savePref():86]: Storing: ConfigBuilder_INSULIN_InsulinOrefRapidActingPlugin_Visible:false
12:21:01.551 [main] D/CONFIGBUILDER: [ConfigBuilderPlugin.savePref():82]: Storing: ConfigBuilder_INSULIN_InsulinOrefUltraRapidActingPlugin_Enabled:false
12:21:01.551 [main] D/CONFIGBUILDER: [ConfigBuilderPlugin.savePref():86]: Storing: ConfigBuilder_INSULIN_InsulinOrefUltraRapidActingPlugin_Visible:false
12:21:01.551 [main] D/CONFIGBUILDER: [ConfigBuilderPlugin.savePref():82]: Storing: ConfigBuilder_INSULIN_InsulinLyumjevPlugin_Enabled:false
12:21:01.552 [main] D/CONFIGBUILDER: [ConfigBuilderPlugin.savePref():86]: Storing: ConfigBuilder_INSULIN_InsulinLyumjevPlugin_Visible:false
12:21:01.552 [main] D/CONFIGBUILDER: [ConfigBuilderPlugin.savePref():82]: Storing: ConfigBuilder_INSULIN_InsulinOrefFreePeakPlugin_Enabled:true
12:21:01.552 [main] D/CONFIGBUILDER: [ConfigBuilderPlugin.savePref():86]: Storing: ConfigBuilder_INSULIN_InsulinOrefFreePeakPlugin_Visible:true
12:21:01.553 [main] D/CONFIGBUILDER: [ConfigBuilderPlugin.savePref():82]: Storing: ConfigBuilder_SENSITIVITY_SensitivityAAPSPlugin_Enabled:false
12:21:01.553 [main] D/CONFIGBUILDER: [ConfigBuilderPlugin.savePref():86]: Storing: ConfigBuilder_SENSITIVITY_SensitivityAAPSPlugin_Visible:false
12:21:01.553 [main] D/CONFIGBUILDER: [ConfigBuilderPlugin.savePref():82]: Storing: ConfigBuilder_SENSITIVITY_SensitivityWeightedAveragePlugin_Enabled:false
12:21:01.553 [main] D/CONFIGBUILDER: [ConfigBuilderPlugin.savePref():86]: Storing: ConfigBuilder_SENSITIVITY_SensitivityWeightedAveragePlugin_Visible:false
12:21:01.553 [main] D/CONFIGBUILDER: [ConfigBuilderPlugin.savePref():82]: Storing: ConfigBuilder_SENSITIVITY_SensitivityOref1Plugin_Enabled:true
12:21:01.554 [main] D/CONFIGBUILDER: [ConfigBuilderPlugin.savePref():86]: Storing: ConfigBuilder_SENSITIVITY_SensitivityOref1Plugin_Visible:true
12:21:01.554 [main] D/CONFIGBUILDER: [ConfigBuilderPlugin.savePref():82]: Storing: ConfigBuilder_PUMP_DanaRPlugin_Enabled:false
12:21:01.554 [main] D/CONFIGBUILDER: [ConfigBuilderPlugin.savePref():86]: Storing: ConfigBuilder_PUMP_DanaRPlugin_Visible:false
12:21:01.554 [main] D/CONFIGBUILDER: [ConfigBuilderPlugin.savePref():82]: Storing: ConfigBuilder_PUMP_DanaRKoreanPlugin_Enabled:false
12:21:01.555 [main] D/CONFIGBUILDER: [ConfigBuilderPlugin.savePref():86]: Storing: ConfigBuilder_PUMP_DanaRKoreanPlugin_Visible:false
12:21:01.555 [main] D/CONFIGBUILDER: [ConfigBuilderPlugin.savePref():82]: Storing: ConfigBuilder_PUMP_DanaRv2Plugin_Enabled:false
12:21:01.555 [main] D/CONFIGBUILDER: [ConfigBuilderPlugin.savePref():86]: Storing: ConfigBuilder_PUMP_DanaRv2Plugin_Visible:false
12:21:01.555 [main] D/CONFIGBUILDER: [ConfigBuilderPlugin.savePref():82]: Storing: ConfigBuilder_PUMP_DanaRSPlugin_Enabled:true
12:21:01.556 [main] D/CONFIGBUILDER: [ConfigBuilderPlugin.savePref():86]: Storing: ConfigBuilder_PUMP_DanaRSPlugin_Visible:true
12:21:01.556 [main] D/CONFIGBUILDER: [ConfigBuilderPlugin.savePref():82]: Storing: ConfigBuilder_PUMP_LocalInsightPlugin_Enabled:false
12:21:01.556 [main] D/CONFIGBUILDER: [ConfigBuilderPlugin.savePref():86]: Storing: ConfigBuilder_PUMP_LocalInsightPlugin_Visible:false
12:21:01.556 [main] D/CONFIGBUILDER: [ConfigBuilderPlugin.savePref():82]: Storing: ConfigBuilder_PUMP_ComboPlugin_Enabled:false
12:21:01.556 [main] D/CONFIGBUILDER: [ConfigBuilderPlugin.savePref():86]: Storing: ConfigBuilder_PUMP_ComboPlugin_Visible:false
12:21:01.557 [main] D/CONFIGBUILDER: [ConfigBuilderPlugin.savePref():82]: Storing: ConfigBuilder_PUMP_MedtronicPumpPlugin_Enabled:false
12:21:01.557 [main] D/CONFIGBUILDER: [ConfigBuilderPlugin.savePref():86]: Storing: ConfigBuilder_PUMP_MedtronicPumpPlugin_Visible:false
12:21:01.557 [main] D/CONFIGBUILDER: [ConfigBuilderPlugin.savePref():82]: Storing: ConfigBuilder_PUMP_DiaconnG8Plugin_Enabled:false
12:21:01.557 [main] D/CONFIGBUILDER: [ConfigBuilderPlugin.savePref():86]: Storing: ConfigBuilder_PUMP_DiaconnG8Plugin_Visible:false
[... cut ...] 7,97,97,96,96,96,96,96,96,96,96,96,96,95],"ZT":[120,118,116,115,113,111,110,108,107,106,105,103,102,101,101,100,99,98,98,97,96,96,95,95,94,94,94,93,93,93,93,92,92,92,92,92,92,91,91,91,91,91,91,91,91,91,91,91],"UAM":[120,119,118,116,115,114,112,111,110,108,107,106,105,104,103,102,101,101,100,99,99,98,98,97,97,97,96,96,96,95,95,95,95,95,94,94,94,94,94,94,94,94,94,94,94,93]},"COB":0,"IOB":0.216,"reason":"COB: 0, Dev: 5, BGI: -2, ISF: 132, CR: 13.4, Target: 85, minPredBG 94, minGuardBG 93, IOBpredBG

MTR93600, 2021-09-27:
if I have to make a focus on the log here you have the part with the settempbasalpercent failed : 

12:21:55.511 [Thread-791] D/PUMPBTCOMM: [BLEComm.connect():95]: Initializing BLEComm.
12:21:55.512 [Thread-791] D/PUMPBTCOMM: [BLEComm.connect():130]: Trying to create a new connection from: Connection needed
12:21:55.521 [Thread-791] D/PUMPBTCOMM: [BLEComm.setCharacteristicNotification():280]: setCharacteristicNotification
12:21:56.031 [main] D/CONSTRAINTS: [StorageConstraintPlugin.isClosedLoopAllowed():40]: Internal storage free (Mb):115880
12:21:56.032 [main] D/CONSTRAINTS: [StorageConstraintPlugin.isClosedLoopAllowed():40]: Internal storage free (Mb):115880
12:21:56.036 [main] D/CORE: [OverviewFragment.updateGUI():718]: Adjusted target. Profile: 90.0 APS: 85.0
12:21:56.521 [Thread-791] D/PUMPQUEUE: [QueueThread.run():95]: connecting 1
12:21:56.671 [Binder:1261_6] D/PUMPBTCOMM: [BLEComm.onConnectionStateChangeSynchronized():358]: onConnectionStateChange
12:21:56.674 [Binder:1261_5] D/PUMPBTCOMM: [BLEComm$mGattCallback$1.onServicesDiscovered():237]: onServicesDiscovered
12:21:56.674 [Binder:1261_5] D/PUMPBTCOMM: [BLEComm.getSupportedGattServices():326]: getSupportedGattServices
12:21:56.675 [Binder:1261_5] D/PUMPBTCOMM: [BLEComm.setCharacteristicNotification():280]: setCharacteristicNotification
12:21:56.931 [Binder:1261_5] D/PUMPBTCOMM: [BLEComm.sendConnect():517]: >>>>> ENCRYPTION__PUMP_CHECK (0x00) A5 A5 0C DF  F2 DC 9B B7  B8 EE C3 B8  EF B4 CA 4A  45 5A 5A 
12:21:57.157 [Binder:1261_5] D/PUMPBTCOMM: [BLEComm.processConnectResponse():549]: <<<<< ENCRYPTION__PUMP_CHECK V3 (OK) 02 00 4F 4B  4D 05 50 11  18 
12:21:57.158 [Binder:1261_5] D/PUMPBTCOMM: [BLEComm.sendV3PairingInformation():644]: >>>>> ENCRYPTION__TIME_INFORMATION A5 A5 03 DF  F3 88 DE B3  5A 5A 
12:21:57.291 [Binder:1261_5] D/PUMPBTCOMM: [BLEComm.processEncryptionResponse():650]: <<<<< ENCRYPTION__TIME_INFORMATION 02 01 00 
12:21:57.291 [Binder:1261_5] D/PUMPBTCOMM: [BLEComm.processEncryptionResponse():664]: Connect !!
12:21:57.522 [Thread-791] D/PUMPQUEUE: [QueueThread.run():110]: connection time 2s
12:21:57.523 [Thread-791] D/PUMPQUEUE: [QueueThread.run():116]: performing TEMP BASAL 0% 15 min
12:21:57.524 [Thread-791] D/CONSTRAINTS: [Constraint.set():28]: Setting value 0 -> 0 (Limiter le pourcentage max à 0% à cause de Limite de la pompe)[Safety]
12:21:57.525 [Thread-791] D/CORE: [TemporaryBasalStorage.add():21]: Stored temporary basal info: TemporaryBasal(timestamp=1632738117524, duration=900000, rate=0.0, isAbsolute=false, type=EMULATED_PUMP_SUSPEND, id=0, pumpId=0)
12:21:57.525 [Thread-791] D/PUMPCOMM: [DanaRSPacketBasalSetCancelTemporaryBasal.<init>():13]: Canceling temp basal
12:21:57.526 [Thread-791] D/PUMPBTCOMM: [BLEComm.sendMessage():771]: >>>>> BASAL__CANCEL_TEMPORARY_BASAL B2 62  
12:21:57.876 [Binder:1261_5] D/PUMPBTCOMM: [BLEComm.processMessage():851]: <<<<< BASAL__CANCEL_TEMPORARY_BASAL B2 62 00 
12:21:57.877 [Binder:1261_5] D/PUMPCOMM: [DanaRSPacketBasalSetCancelTemporaryBasal.handleMessage():19]: Result OK
12:21:58.377 [Thread-791] D/PUMPCOMM: [DanaRSPacketAPSBasalSetTemporaryBasal.<init>():18]: New message: percent: 0
12:21:58.378 [Thread-791] D/PUMPCOMM: [DanaRSPacketAPSBasalSetTemporaryBasal.<init>():25]: APS Temp basal start percent: 0 duration 30 min
12:21:58.378 [Thread-791] D/PUMPBTCOMM: [BLEComm.sendMessage():771]: >>>>> BASAL__APS_SET_TEMPORARY_BASAL B2 C1  00 00 A0 
12:21:58.491 [main] D/info.nightscout.androidaps.utils.ui.SingleClickButton: [SingleClickButton.guardClick$lambda-0():17]: Button enabled
12:21:58.677 [Binder:1261_5] D/PUMPBTCOMM: [BLEComm.processMessage():851]: <<<<< BASAL__APS_SET_TEMPORARY_BASAL B2 C1 00 
12:21:58.678 [Binder:1261_5] D/PUMPCOMM: [DanaRSPacketAPSBasalSetTemporaryBasal.handleMessage():47]: Set APS temp basal start result: 0
12:21:58.684 [OverviewFragmentHandler] D/UI: [OverviewPlugin.refreshLoop():247]: refreshLoop finished
12:21:59.680 [Thread-791] D/PUMPCOMM: [DanaRSPacketAPSHistoryEvents.<init>():46]: Loading event history from: 9/27/21 12:20
12:21:59.681 [Thread-791] D/PUMPCOMM: [DanaRSService.loadEvents():230]: Loading event history from: 9/27/21 12:20
12:21:59.682 [Thread-791] D/PUMPBTCOMM: [BLEComm.sendMessage():771]: >>>>> APS_HISTORY_EVENTS B2 C2  15 09 1B 0C  14 17 
12:22:00.000 [Binder:1261_5] D/PUMPBTCOMM: [BLEComm.processMessage():851]: <<<<< APS_HISTORY_EVENTS B2 C2 01 15  09 1B 0C 16  01 00 00 00  1E 
12:22:00.024 [Binder:1261_6] D/PUMPBTCOMM: [BLEComm.processMessage():851]: <<<<< APS_HISTORY_EVENTS B2 C2 02 15  09 1B 0C 16  01 00 00 00  00 
12:22:00.082 [Binder:1261_6] D/PUMPBTCOMM: [BLEComm.processMessage():851]: <<<<< APS_HISTORY_EVENTS B2 C2 05 15  09 1B 0C 15  17 00 05 00  00 
12:22:00.146 [Binder:1261_5] D/PUMPBTCOMM: [BLEComm.processMessage():851]: <<<<< APS_HISTORY_EVENTS B2 C2 FF 03  
12:22:00.147 [Binder:1261_5] D/PUMPCOMM: [DanaRSPacketAPSHistoryEvents.handleMessage():78]: Last record received
12:22:00.148 [Binder:1261_5] D/PUMP: [DetailedBolusInfoStorage.findDetailedBolusInfo():56]: Bolus info not found for time 1632738083000
12:22:00.156 [Binder:1261_5] D/PUMPCOMM: [DanaRSPacketAPSHistoryEvents.processMessage():188]: [1632738083000] EVENT BOLUS (5) 9/27/21 12:21 (1632738083000) Bolus: 0.05U 
12:22:00.157 [Binder:1261_5] D/PUMP: [TemporaryBasalStorage.findTemporaryBasal():32]: Using & removing temporary basal info: TemporaryBasal(timestamp=1632738117524, duration=900000, rate=0.0, isAbsolute=false, type=EMULATED_PUMP_SUSPEND, id=0, pumpId=0)
[... cut ...]
12:22:00.329 [Thread-791] D/PUMPQUEUE: [QueueThread.run():138]: waiting for disconnect
12:22:00.331 [main] D/CORE: [AlarmSoundService.onCreate():62]: onCreate parent called
12:22:00.343 [main] D/CORE: [AlarmSoundService.onCreate():64]: onCreate End
12:22:00.361 [main] D/CORE: [AlarmSoundService.onStartCommand():68]: onStartCommand
12:22:00.363 [main] D/CORE: [AlarmSoundService.onStartCommand():70]: onStartCommand Foreground called
12:22:00.420 [main] D/CORE: [AlarmSoundService.onStartCommand():96]: onStartCommand End

MilosKozak, 2021-09-28:
i tested everything again and it works for me.
Anyway I added more logging to this part of code.
Please try again with moste recent dev and provide logs again

MTR93600, 2021-11-02:
Hi,
I'm using the beta 2, and I still have the TBR pb, with most of the time no smb. I'm using Dana RS. I don't have this issue on the master. I know you didn't reproduce the pb, and it's not like I never receive smb, but the false come to often to ovoid hyperglycemia during the post meal time. I look in the code and can not find anything related with this...  
![photo_2021-11-02_17-55-46](https://user-images.githubusercontent.com/44895345/139910474-0c1e871e-21f5-4f6d-8063-687a5eb31137.jpg
[log.log](https://github.com/nightscout/AndroidAPS/files/7462628/log.log)
)

MTR93600, 2021-11-02:
[log.log](https://github.com/nightscout/AndroidAPS/files/7462643/log.log)

MTR93600, 2021-11-02:
[image]

MilosKozak, 2021-11-02:
this time setting of TBR really failed (reported by pump)
as this usually works, i'd try update the firmware in pump

MTR93600, 2021-11-03:
i have the V3, I didn't heard about an update. And more importantly, new firmware will work with aaps ? can you confirm which version you are using please

MilosKozak, 2021-11-03:
you may try this https://play.google.com/store/apps/details?id=com.cozmo.anydana.upgradekit&hl=cs

MTR93600, 2021-11-03:
I did that already to get the v3 when you update the drive in the aaps master 2.8.2

CaroGo, 2021-11-06:
I observed the same issue.... Looks like this... 
[image]

MilosKozak, 2021-11-10:
any new logs with issue 
beta5 or newer

MTR93600, 2021-11-12:
here the log, last update
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/7530111/AndroidAPS.log)

MTR93600, 2021-11-24:
hi Mylos, i have again the tbr/smb pb today
here the lo
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/7595028/AndroidAPS.log)
g

MilosKozak, 2021-11-24:
hopefully resolved by https://github.com/nightscout/AndroidAPS/commit/0dbfa6182f996674c3ab72aa1fc19684d8cc00ae

closed 2021-11-24 by MilosKozak