nightscout/AndroidAPS #693: 2.8.2.3 dev Automation does not color profile %

robertrub, 2021-09-26:
More od an infirmation mini bug.
When an automation kicks in a _profile %_ and _TT_ change, in 2.8.2 master, both informations are in yellow 

[image]

In 2.8.2.3 dev (build a55eb0623) the profile % stays black.

[image]

Philoul, 2021-09-26:
I can take a look into the code for the missing yellow background 😉

MilosKozak, 2021-10-03:
resolved by https://github.com/nightscout/AndroidAPS/commit/214078b47740ce1b7e37be11c30ef9c470103e26

closed 2021-10-03 by MilosKozak