nightscout/AndroidAPS #706: Alarm dialog gone after minimizing and maximizing the app

CaptainBalou, 2021-10-02:
### Description
After issues with delivering a bolus an alarm was triggered. I could hear the audio alarm and I could see the alarm dialog which offered me to mute the alarm, to snooze the alarm and one option more.
Unfortunately I pressed the "minimize the application" home button of my phone.
After maximizing the application the audio was still there but the alarm dialog was gone. So I couldn't stop the alarm. I had to reboot the device to get rid of it.

### Expected behaviour
The alarm dialog is there also after minimizing and again maximizing the application to be able to stop the alarm.

### Version
Version: 2.8.2.3-dev
Build: b2c45d0a2-2021.09.27-00:36

MilosKozak, 2021-10-03:
minimizing works properly. you had to close the app. if this case there is no solution ....

closed 2021-10-03 by MilosKozak

CaptainBalou, 2021-10-03:
In my opinion the behavior should be that the alarm dialog is shown again after maximizing the app as long as the alarm is ongoing. The need to restart the app to quit the alarm sounds weird to me. Like if I would watch a YouTube video and after minimizing the app the video still runs in the background with the audio but I cannot see the video after maximizing the app nor being able to stop it.

MilosKozak, 2021-10-04:
i tested it and for me it works properly