nightscout/AndroidAPS #738: Cannot Open Loop DEV 2.8.2.5

pbloom209, 2021-10-16:
I disconnected the pump as I am aware there is an issue on suspending the Loop.

Then proceeded to change reservoir and site.  It took about ten mins.

When complete the Reconnect would not work and it showed the Loop and I could not start it.

I had to reinstall the app.

Running Medtronic 754 
OL

andyrozman, 2021-10-16:
Restarting of phone and OL, should have corrected the problem... 

Which firmware do you have on your OL? I was testing today through day, and it seems that latest version (3.2) doesn't work good. 

I am doing some fixes at the moment on MDT and also some changes on new OL patch that was supplied and merged into dev in last few days... 

How old is your dev? Did you build it before OL fix was merged?

andyrozman, 2021-10-16:
Mhm. Week old would be missing OL changes from bubbledevteam. 

This versions don't tell me anything... All versions of OL show the same version. There are 5 versions of OL: 1.5, 1.7, 2.1, 2.5 and 3.2. So the last version doesn't work that good (at least not with my setup). You can find all information here and you can upgrade it yourself: https://getrileylink.org/orangelink-firmware    

I would recommend that you upgrade OL to 2.5 version.

If you feel adventurous, you can try my branch (my changes will be merged tomorrow, but I am not sure if it will be in time for RC1 release, I did some fixes and also some changes to OL patch): https://github.com/andyrozman/AndroidAPS/tree/mdt_syncstorage