nightscout/AndroidAPS #751: [3.0-beta1] OrangeLink didn't work with Medtronic anymore

Alamo04, 2021-10-18:
OL connected first after phone restart, then OL connect and AAPS tryed to read the Medtronic history.
- History reading started and will interrupt immediatly.
- Message "Problem to connect pump...

Logs sendet to @andyrozman , because it was late and over Phone, before I was going back to working version 2.8.2.5 from 16.10.2021-11:22.

tanja3981, 2021-11-16:
@MilosKozak @andyrozman So do we leave AAPS 3.0 just as it is, working with older firmware only?
Because people who have already updated their OL or got it with the new firmware can't use 3.0 then, can they?

Alamo04, 2021-11-16:
Version 3.2 is the greatest shit for Medtronic AAPS users.
The changes are all made for Omnipod iOS Looper and makes big problems for Medtronic AAPS user.

So using version 2.5 is way better than using 3.2 and as unprofessionell this 3.2 was made, I don't trust any versions in the future and will stay on 2.5 until many reports coming in, that a new firmware was a great improvement.

andyrozman, 2021-11-17:
Version 3.2 has a bug. I talked to developer and he is working on fixing it. If you downgrade to 2.5 it should work with any AAPS version.

There is no error in our (AAPS) code, just in OL firmware itself. I haven't tested it with Omnipod, but it definitely doesn't work for Medtronic.

@Alamo04 
We notified the users that version 3.2 doesn't work... If you look at release comments, from 1st (or 2nd) beta version (on FB) you will find it.