nightscout/AndroidAPS #771: [3.0-beta1] New profile is not correctly shown after profile switch if the new profile has the same content as the old

reijoahola-nightscout, 2021-10-21:
I first noticed this with 3.0-beta1 but actually tested it also with yesterday dev-version and I believe that it still exists.
I cloned earlier profile and edited a new name for it and then saved and activated it but the profile name did not change
in the top of main ´display. The new profile is saved correctly and it also looks like the profile switch is also logged but the display of the profile name does not work. When the content is the same as in earlier profile it does not cause such problems but in the longer term this name-problem can be very confusing.

And for a clarification: if the new profile has something different then there is no problem.

Could somebody check is this a real problem or has I done some mistake.

Philoul, 2021-10-31:
One annoying side effect is for new users that switch from current master to new beta, they first have to activate a new profile to allow looping, but if they try to activate the same profile that the one recorded in pump (it's generally the case), they still have "No profile set" in overview after their profile switch (my advice is to do a PS with xx% and then another one with 100%, it works but it's a workaround difficult to understand by most users...)

reijoahola-nightscout, 2021-11-03:
Tested the same situation with 3.0-beta3: Active Profile copied into a new name and then activated. Now the new profile is shown in the top of the main view. Is the logic then changed ?

MilosKozak, 2021-11-03:
yes, i changed the behavior the way that change name is a reason for update too

markvader, 2021-11-03:
[3.0-beta3] I made a change to my profile, the only change i made was one of my ISF rates, then I activated it and completed a profile switch.

The profile shows as changed in the profile plugin and uploads to correct values to nightscout

However the profile is not updated in the profile bar on the home screen, nor is the change taken into account for calculations, and no profile change is shown on the main graph as it normally would.

[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/7466945/AndroidAPS.log)

I have attached the log.  I set the ISF from 7am to 10pm as 9mmol  

Line 38 shows the new rate
`09:47:16.997 [main] D/PROFILE: [LocalProfilePlugin.storeSettings():211]: Storing settings: {"defaultProfile":"LocalProfile","startDate":"2021-11-03T09:47:16.9960000Z","store":{"LocalProfile":{"dia":6,"carbratio":[{"time":"00:00","timeAsSeconds":0,"value":18},{"time":"11:00","timeAsSeconds":39600,"value":14.5},{"time":"16:00","timeAsSeconds":57600,"value":18}],"sens":[{"time":"00:00","timeAsSeconds":0,"value":8},{"time":"07:00","timeAsSeconds":25200,"value":9},{"time":"22:00","timeAsSeconds":79200,"value":8}],"basal":[{"time":"00:00","timeAsSeconds":0,"value":0.55},[other rates removed to reduce log],{"time":"23:00","timeAsSeconds":82800,"value":0.66}],"target_low":[{"time":"00:00","timeAsSeconds":0,"value":6},{"time":"08:00","timeAsSeconds":28800,"value":5.6}],"target_high":[{"time":"00:00","timeAsSeconds":0,"value":6},{"time":"08:00","timeAsSeconds":28800,"value":5.6}],"units":"mmol","timezone":"Europe\/London"}}}
`
Line 135 shows the old rate of 153 (i.e. 8.5mmol):
`09:50:36.524 [RxCachedThreadScheduler-4] D/APS: [DetermineBasalAdapterSMBJS.invoke():73]: Profile:        {"max_iob":7,"type":"current","max_daily_basal":0.72,"max_basal":2.16,"min_bg":100.8,"max_bg":100.8,"target_bg":100.8,"carb_ratio":18,"sens":153,"max_daily_safety_multiplier":3,"current_basal_safety_multiplier":4,"high_temptarget_raises_sensitivity":false,"low_temptarget_lowers_sensitivity":false,"sensitivity_raises_target":true,"resistance_lowers_target":false,"adv_target_adjustments":false,"exercise_mode":false,"half_basal_exercise_target":160,"maxCOB":120,"skip_neutral_temps":false,"remainingCarbsCap":90,"enableUAM":true,"A52_risk_enable":false,"SMBInterval":3,"enableSMB_with_COB":true,"enableSMB_with_temptarget":true,"allowSMB_with_high_temptarget":true,"enableSMB_always":false,"enableSMB_after_carbs":false,"maxSMBBasalMinutes":120,"maxUAMSMBBasalMinutes":70,"bolus_increment":0.1,"carbsReqThreshold":1,"current_basal":0.7,"temptargetSet":false,"autosens_max":1.2,"out_units":"mmol\/L"}`

Also as an additional issue.  
A short press on the profile bar on the home screen opens the profile overview, long-press brings up the profile switch but the profile overview opens on top of the profile switch and has to be dismissed before you can do a profile switch, should this be this way?

old-square-eyes, 2021-11-06:
I'm still having this issue per #808 I'm not sure why both issues are closed. I have spent hours trying to set a profile. It's not possible for me on any of the betas for NSClient.

Alamo04, 2021-11-06:
One in our german looper group had the same issue and that in the master version.
https://de.loopercommunity.org/t/profilwechsel-schlaegt-fehl/8754

This guy was on AAPS 2.8.2 Master and Medtronic pump with OrangeLink Pro and on this $%#@ new firmware 3.2.

I told him to downgrade the OL Pro to v1.0, the normal OL should be downgrade to version 2.5.
After that, everything works fine again.

So, if you are using OL than be sure to not use Firmware 3.2