nightscout/AndroidAPS #772: No BG delta on Beta1

tjbain, 2021-10-21:
I'm having a persistent issue where BG changes (delta) at the top of the screen are reporting as "0" despite my BG obviously changing. This happens every time I bring up the app (app does run in the background, so I'm not talking about a fresh start). After a while, maybe 5 minutes (probably the next BG reading), the delta will show. 
Here's my info:
Samsung Galaxy Note 20 5g
AAPS 3.0-beta1
Omnipod Dash (communication and commands work great!)
[image]

Logs: Ironically, just as I report this, it seems all is okay. So I don't have a log that shows the error at this moment. Will upload ASAP!

MilosKozak, 2021-10-21:
paste screenshots of BG tab

tjbain, 2021-10-21:
> Will absolutely do whatever you guys ask as soon as I get back home and
> have a chance to do this stuff
>
> Thanks for looking into it! I'm sure it's nothing major...probably user
> error

Here's a log I managed to produce.